GlasgowEmbedded/glasgow #33: Develop a tool to validate capacitor/resistor values against Mouser API

whitequark, 2018-05-03:


shapr, 2018-05-10:
could this be unpacked with more detail? Might be able to use the octopart API

whitequark, 2018-05-10:
So, I've had an issue several times that the Mouser part number didn't match the value of a resistor (less a problem with capacitors because we only have like one capacitor whose value actually matters, the rest is decoupling). It's a thing that's *really* easy to miss with manual review, so I want to automate it.

Octopart probably works too, just cut `\d+-` from the start of Mouser PN to get MPN.

whitequark, 2018-05-25:
I'm dumb and this is not necessary. The Yageo part numbers have the resistor value in the part number, e.g. `10K` in `RC0603FR-1310KL` or `604R` in `RC0603FR-07604RL`. Same for package. A cursory visual check is well enough.

closed 2018-05-25 by whitequark