GlasgowEmbedded/glasgow #35: Slew rate limiting resistors

whitequark, 2018-05-07:
Two arrays of four resistors per port. I'm not sure yet what the values should be; worst case, we can always populate with 0R and decide later.

whitequark, 2018-05-07:
This will also add some very limited pin overcurrent protection. Say at 200 ohm, the buffers should tolerate ±10V on up to two signal pins (not Vio or Vsense though).

whitequark, 2018-05-07:
0402 8-position array fits better (it's about the same size as FXMA108).

whitequark, 2018-06-06:
Pretty sure we also need slew rate limiting on the *other* side of the level shifter because switching many inputs at once causes [glitches](https://github.com/whitequark/Glasgow/blob/a89f5a224449ddd5d1bf9e5f4e8b7e3bc2593bce/software/glasgow/gateware/fx2.py#L141-L145) on the FLAGn pins on the other side of the board. The apparently random timing of the glitches seems to suggest that glitches happen when the inputs transition happens closely enough to IFCLK transition.

awygle, 2018-06-25:
I added these in commit 7c3203c126e7444981d5405d7d6fcd1db53bbbc5

commit c9c834d0 (2018-05-12): Add slew rate control resistors to DUT busses.

Closes #35.
closed 2018-07-21 by whitequark