GlasgowEmbedded/glasgow #38: Use an FX2LP in a BGA package?

whitequark, 2018-05-08:
We can save a dollar if we go for 56-ball 0.5 mm pitch VFBGA package. We only need to route out two rows of balls at most, so it seems fine that it's 0.5 mm pitch. Should we?

awygle, 2018-05-08:
Mmm... maybe. The problem with the 0.5mm pitch is, you can't get vias *or* traces between balls on 6/6 trace/space specs. I'll investigate this at the same time as doing the HX routing - 0.8mm parts have similar-but-not-the-same problems, so seeing what we *need* to do will help us figure out what it'll cost us to do the optional stuff.

whitequark, 2018-10-06:
@awygle Do you remember what was your conclusion? Looking at the package, some B and G balls do present an issue...
[image]

whitequark, 2018-10-31:
We don't have any margin for two-row escape routing on 0.8 mm with 5/5 rules already, so this is not viable.

closed 2018-10-31 by whitequark