GlasgowEmbedded/glasgow #49: Implement MPSSE subtarget and FTDI emulation mode

whitequark, 2018-05-13:
This is kind of annoying because it needs reenumeration logic in the firmware.

marcan, 2019-05-01:
Don't forget to emulate the 32-bit EEPROM correctly to avoid being hit by FTDI's bricker :P

whitequark, 2019-08-19:
Does anyone actually *want* this?

esden, 2019-08-20:
MPSSE would be fairly useful. There is a bunch of software that uses MPSSE for "stuff" and it would be nice to just transparently use the Glasgow instead of an FTDI cable. Also that implementation might be a useful reference implementation for FTDI chip replacements.

classic-gentleman, 2020-07-08:
> MPSSE would be fairly useful. There is a bunch of software that uses MPSSE for "stuff" and it would be nice to just transparently use the Glasgow instead of an FTDI cable. Also that implementation might be a useful reference implementation for FTDI chip replacements.

YES, for every point made.

whitequark, 2020-07-08:
Ironically this was the original goal of the project and the very first committed gateware, but I don't think we even have the code left for it in-tree anymore.

classic-gentleman, 2020-07-08:
https://github.com/usb-tools/Facedancer/blob/master/facedancer/devices/ftdi.py

Adapting this shouldn't be horrible...

whitequark, 2020-07-08:
The problem is actually the firmware, and there is no great solution that I know of, especially for Windows.

whitequark, 2023-07-23:
We decided not to do this. 60d09a217502f469682e3dd1c51c5e6bd2dfd586

closed 2023-07-23 by whitequark