GlasgowEmbedded/glasgow #50: Figure out exactly what happens during Vio short-circout to GND

whitequark, 2018-05-13:
On revA the fuse blows immediately. Need to check if the current limiting works as designed (probably not since the fuse blows. lol.) and decide what to do about it.

awygle, 2018-05-22:
I put a hard short on one VIO port and did not pop the fuse. I measured a peak of 350 mA and 280 mA steady state (without the short and with the I/O toggling at 15 MHz I was seeing 77 mA). I put a hard short on the other port as well and managed to pop it. That suggests to me that we're close to being able to tolerate two full shorts on the I/O banks, and that dropping down to 100 mA LDOs would probably put us into safe territory. It also becomes sort of a question as to whether we need the USB IC at that point... But then you had your issue earlier so I'm not totally sure what's going on.

Thoughts? Can you replicate your one-port fuse popping?

awygle, 2018-05-22:
With both shorts in place I nearly popped the 600 mA fuse on my ammeter - it looked like 580 mA with both LDOs in current limit.

With one short in place I got up to 350 mA and then it looked like the LDO was going into a thermal shutdown loop. So I think the full-short current is pretty much 350 mA. But again, I was only able to get 77 mA from the pins toggling at 15 MHz, so it looks like the LDO is actually limiting much higher than 150 mA.

awygle, 2018-05-22:
I am an idiot. The current limit for the TPS731 is min 150, typ 360, max 500 mA. No wonder it's not protecting us.

I was going to suggest the LP2980-ADJ, which is a 50 mA output LDO, and whose current limit is "typical" 150 mA. That would put us into safe territory. I'd be worried about starving the shifters but it looks like we can't feed them any faster than like 30 MHz, so we should be safe.

whitequark, 2018-05-22:
> Thoughts? Can you replicate your one-port fuse popping?

I, uh, don't have any fuses left.

whitequark, 2018-05-22:
> I'd be worried about starving the shifters but it looks like we can't feed them any faster than like 30 MHz, so we should be safe.

Hm, I wouldn't put 50 MHz out of question with async FIFOs.

awygle, 2018-05-22:
Yeah, the dropout's not ideal. The alternative is to rely on the USB protection IC. The bummer with that is that it will probably brown out the FX2 and cause the device to flap on the user's PC. The LDO will stop this further downstream.

whitequark, 2018-05-22:
I think a brownout is OK.

whitequark, 2018-05-25:
Since a brownout seems OK (and even beneficial in some ways, as you get a pretty visible indication of something going wrong!) I'm closing this as a duplicate of #34.

closed 2018-05-25 by whitequark