GlasgowEmbedded/glasgow #51: Make SYNC pullup smaller

whitequark, 2018-05-14:
The FPGA has ridiculously powerful drivers (what, 10 mA?) so if we use a pullup smaller than 1k we can go well into MHz range. Useful, because the PLL has a minimum frequency of 16 MHz--can we get that high?

awygle, 2018-05-22:
I think 300 Ohms is the lowest we can safely go. That'll give us 300 mV of rise above ground at 3.3V with 10 mA drivers.

whitequark, 2018-05-22:
It's actually 24 mA. On the other hand, Vil(max) for LVCMOS 3.3 inputs is 0.8 V, so 0.3 V seems fine.

awygle, 2018-05-22:
Okay, that'd give us 125 Ohms. 130 for ease of sourcing. Math math math.

whitequark, 2018-05-22:
Can you verify that this value actually gives a decent signal? I don't have my resistor set at hand...

awygle, 2018-05-22:
I have some 120 Ohm resistors, I can look at what those give.

awygle, 2018-06-25:
Where'd you find 24 mA as the value? I'm looking at 
[image] and reading 8 mA.

(Just for completeness that's 375 Ohms by math and probably 390 by sourcing)

whitequark, 2018-07-10:
Hrm, I might have been looking at the wrong package. 300 mV rise seems pretty low too, that's not enough to sync using e.g. external LMVCMOS 3.3 circuitry... Let's aim at 800 mV rise above ground then.

awygle, 2018-07-10:
Because it's an open drain output, 300 mV is Voh in this case - the voltage for a logic 0. The high voltage will be the full VIO voltage.

whitequark, 2018-07-10:
Oh right, I derped. That's fine then, let's go with 300 mV.

awygle, 2018-07-22:
Whoops:

[image]

commit 0ecd8ce4 (2018-07-21): Reduce SYNC pull-up value to 390 Ohms.

Closes #51.
commit 3b602a50 (2018-07-22): Further reduce SYNC pullup to 130 Ohms.

Closes #51 (again).
closed 2018-07-22 by awygle