GlasgowEmbedded/glasgow #59: ESD protection on Vio and Vsense pins

whitequark, 2018-05-21:
FXMA108 gives us ESD protection on all I/O pins. However, what about the ADC and the LDO? In particular, the turned off LDO has the output in hi-Z state, which seems like it might have bad implications. Verify.

awygle, 2018-05-22:
The LDO has 2 kV ESD tolerance, the ADC has 2.5 kV (on the pins in question, under the human body model). The LDO rail has 4.7 uF of capacitance, and the ADC has 12 nF before the input pin. Given that the human body model specifies 200 pF of capacitance, we have an effective 250 kV of tolerance, which should be plenty.

See https://irclog.whitequark.org/~h~openfpga/2018-05-22#22149898; for discussion with @azonenberg on this topic.

whitequark, 2018-05-22:
Agreed.

closed 2018-05-22 by whitequark