GlasgowEmbedded/glasgow #67: Add pullups/pulldowns on every pin

whitequark, 2018-10-07:
I thought about this a lot and pullups are *everywhere*. Having zero external component pullups is very important.

I suggest using PCF8574RGT plus a 4- or 8-wide resistor network to get tristate/pullup/pulldown capacity on every pin. It will have to be mounted on the bottom side and complicate routing even more, but it's worth the hassle.

Bikeshed: what should we make the pull value? 10k sounds good to me.

hardkrash, 2018-10-15:
Bikeshed... Red My favorite color.
- Nice pull values are between 3.3K for strong i2c pull-ups to 100k for weak drive chips.
- Consider a few channels with individual 0603 resistor or small surface mount 100k trimpot for easy changes.
  - e.g. Bournes TC33X-2-104E in series with 2.2k to 3.3k resistor to limit lower bound.

whitequark, 2018-10-15:
> Consider a few channels with individual 0603 resistor or small surface mount 100k trimpot for easy changes.
> * e.g. Bournes TC33X-2-104E in series with 2.2k to 3.3k resistor to limit lower bound.

I think I'll go with 5k or 10k for revC. I think further revs might use a current source instead of a pullup so you can use a current mirror to set their strength, but I'll need to find a suitable IC and/or make a larger board. I think revC board just doesn't have enough space for configurable pulls.

hardkrash, 2018-10-15:
i thought about a DAC + current mirror,  but didn't think of how to overcome compliance voltages with single supply 5V, unless this is not a concern with pulls.

It appears that digital-pots are not too expensive these days.  Something like a MCP4241 might work for less than $0.76 @ 5k qty with 8x per board, so $6.08 per board...

hardkrash, 2018-10-15:
Yea board space is always a problem.  I was not necessarily saying the digi-pots for rev C.
An add-on daughter card to have the digital-pots that mimics cordwood type interconnects using longer tails on the interconnects.  Any spare pins on the FPGA? It looked full on the schematic.

It the options are 5k or 10k, a user can always add a resistor externally if need be.

whitequark, 2018-10-15:
> Any spare pins on the FPGA? It looked full on the schematic.

Not on this one, but UP5K can't support revC anyway, so we're switching to HX8K.

> An add-on daughter card to have the digital-pots that mimics cordwood type interconnects using longer tails on the interconnects.

Not a bad idea! I should add some sort of header for I2C so you can hook up other stuff to the bus.

whitequark, 2018-11-22:
Mostly done in master, still needs level shifters for the TCAs.

marcan, 2018-11-25:
Done. What I ended up doing is adding 1.27mm female pin headers around the resistor network for the pull-ups. The user can then opt to desolder the resistor network and plug in their own resistors (discretes, daughtercard, or whatever).

closed 2018-11-25 by marcan