GlasgowEmbedded/glasgow #96: Level shifter state during reconfiguraiton?

whitequark, 2019-01-28:
It looks like we are constrained in level shifter routing: the DIR input is referenced to VCCA. When DIR is low, B is an input. Ideally, we would have this as the state of the tool when the FPGA is reconfigured, but AFAIK the FPGA pins are all set to be weak pullups.

Should we add external pulldowns? This seems... basically impossible placement-wise.

The alternative is to force the entire level shifter into a high impedance state by disabling the voltage regulators. (Good thinking, @marcan--if we powered level shifters directly from Vsense this would not be possible.) The problem is that discharging the output capacitor below the cutoff level may take a while and we cannot directly determine when this happens. So it would be a hardcoded worst-case delay in the FPGA reconfiguration code.

Thoughts?

marcan, 2019-01-28:
Pull-downs are plausible on the back side. It won't be fun but I think it's doable. We already have a couple of resistors back there.

Our only static load on VIO is the feedback divider, which is 83kΩ. We have about 6µF of capacitance on that bus, so not considering any external capacitance, that's a t<sub>RC</sub> of half a second. Not terrible, not great. The easiest change to improve this would be to dump a 1kΩ discharge resistor on there (easy to retrofit to revC0 units), which would add a worst-case load of 5mA and reduce the t<sub>RC</sub> to 6ms. This is probably a good idea anyway, we'd want level changes on that rail to be responsive.

marcan, 2019-01-28:
I couldn't find a proper threshold for the shifters to shut down, but I'm guessing 3t<sub>RC</sub> will probably do it in all cases (that's 5V to 0.23V).

whitequark, 2019-02-02:
Discharge resistor sounds great, and we need that regardless of this issue--I think I've hit the problem with rails changing too slowly before.

We can qualify the level shifter threshold easily, at least the ballpark.

marcan, 2019-03-28:
Let's call this closed for revC1. Discharge resistors are there, we can revisit and add pull-downs if it becomes a problem.

closed 2019-03-28 by marcan