GlasgowEmbedded/glasgow #106: Pulls on revC0 are BROKEN and I²C level shifters require replacement

whitequark, 2019-03-01:
This is what happens to the FX2-side I2C bus after trying to address a pull TCA9534 while *both* DACs are enabled and supply IO power:
[image]
The TCA9534-side I2C bus is similar but oscillates at 1.8 MHz instead.

This happens because we connected two TCA9517s with their B-sides on the same bus. (We had to, since the A-side is the only one that goes below 2.7 V). The datasheet says to not do it, but doesn't mention that they will oscillate wildly...

As far as I can tell, TCA9509 ([datasheet](https://www.ti.com/lit/ds/symlink/tca9509.pdf)) is a pin and package compatible replacement for TCA9517 ([datasheet](https://www.ti.com/lit/ds/symlink/tca9509.pdf)) that should not have this problem. It is possible to use it on existing revC0 boards by swapping the shifter and changing R36-R39 to DNP.

TCA9509 datasheet explicitly calls out this problem (or lack thereof):

> The output pull-down on the B-side drives a hard LOW and the input level is set at 0.3 of SMBus
or I2C-bus voltage level which enables B side to connect to any other I2C-bus devices or buffer.

marcan, 2019-03-08:
Fixed with the new expanders, just need to update the design and validate. Chips are on the way.

marcan, 2019-03-28:
New chips validated and design updated.

closed 2019-03-28 by marcan