GlasgowEmbedded/glasgow #107: Consider connecting level shifter EN to DAC EN

whitequark, 2019-03-01:
Related to #96. We really don't need the level shifters to guess whether the IO port is enabled, and lock up the bus if it guesses wrong.

TCA9509 does this:

> The TCA9509 drivers are not enabled unless VCCA is above 0.8 V and VCCB is above 2.5 V. The enable (EN) pin can also be used to turn the drivers on and off under system control. Caution should be observed to only change the state of the EN pin when the bus is idle.

which is good enough, but I think explicitly managing EN would be safer.

marcan, 2019-03-08:
Sounds good to me. We still need to make sure we have something like a 100ms delay after Vio shutdown before reconfiguration to allow for plenty of time for charge to drain on revC0, but this makes revC1 more robust in the face of external capacitance.

marcan, 2019-03-28:
Kill it with fiiiiiiiire

closed 2019-03-28 by marcan