GlasgowEmbedded/glasgow #117: Pins for Addon-Boards

electroniceel, 2019-03-31:
Let's say I wanted to develop an addon board like suggested as solution for #115. It would make developing an analog addon board easier if I had access to these signals:
5V, SCL, SDA. 
Maybe also one of the aux signals would be handy if an additional pin is free.

These are already accessible on the pcb as smd pads. But that makes bringing them up to an addon board sitting on top and plugging into port A & B mechanically difficult. So I propose to provide these signals on some vertical pin headers that an addon board could easily plug into.

I propose to use the space below the Sync-header for this. In the current revC design this place seems to be unused on most layers and the signals mentioned are already running near that place. The sync header would have to be replaced by a vertical 3x2 pin header. This way it could still be used with pin sockets like in the current design.

The addon board could route through the sync signal and provide it on a pin header again, so you could use the sync with the addon board installed.

What do you think?

whitequark, 2019-03-31:
Duplicate of #68, please continue the discussion there. In short, this wouldn't be possible on revC1, but will likely be possible on revD0.

closed 2019-03-31 by whitequark

whitequark, 2019-04-01:
To be clear--I would very much like to see an analog frontend that works with revC1. I think it should use the normal data pins instead of the internal I²C bus. Indeed that would be a much better experience because you could get higher sample rates and more flexibility with less cost in terms of software complexity, as you don't have to configure the frontend with the FX2.

electroniceel, 2019-04-01:
Sorry for the duplicate, didn't see it because it was closed. Will continue the discussion in #68.