GlasgowEmbedded/glasgow #118: Improve ESD immunity

electroniceel, 2019-04-01:
Glasgow is designed to withstand hotplugging, manually touching wires and so on on the port pins. But there are also other pins accessible from the outside: sync, aux, lvds. They currently aren't protected well. They don't have a series resistor like the ports, so the IC connected to them has to survive all the ESD energy.

The first thing I propose is to add a series resistors (I suggest 47 ohms) between the sync pin and it's '45 U32, and the aux pins and the ice40.

These series resistors help the ICs dropping the ESD voltage. They aren't perfect, but a small, cheap and easy method. In my experience they significantly improve esd immunity. Also they double as a series termination and help to reduce ringing.

The 74LVC1T45s used on the port pins just have an ESD rating of 2 kV HBM. That is not much for a pin connected to outside of the board and expected to be "abused", even with the series resistors in front. I suggest to add TVS diode arrays behind the series resistors to improve this.

My usual go-to part for this application is IP4220CZ6, I use them a lot on my boards. It protects 4 lines of USB HS speed level. But it is SOT23-6, so it wouldn't fit on the board. But there are many similar products available in smaller packages, e.g. D5V0F4U6V, XBP14E5UFN-G or you could even go crazy small with HSP051-4M5.

There are also some available for 8 lines in parallel. These are mostly designed for USB 3, HDMI and the like and could also be used for the LVDS port. For example ESD7016MUTAG,  ESD8008MUTAG or SP7538PUTG.

On the LVDS port I wouldn't add series resistors as this interferes with LVDS and isn't recommended. So just a TVS diode array there.

whitequark, 2019-04-02:
So, the AUX pins and the LVDS bank are not going to be populated as they come from the fab, and moreover, they will not be directly accessible under the case. That means you can't just touch them, and that's why they have no particular protection mechanisms; for LVDS specifically it is expected that for any real use, an add-on board will be connected that has the termination (as it stands, the "LVDS" bank can't really do LVDS, it requires external passives, a lot of them).

Regarding banks A and B and the Sync pin... it would be very tricky to fit. @marcan?

marcan, 2019-04-02:
Adding series resistors for SYNC and AUX is trivial, that much we can do.

As for clamp diodes, it's worth noting that as far as I can tell the SN74LVC1T45s we're using only have low side clamp diodes (because they are 5V tolerant on inputs regardless of IO voltage). So to provide proper protection it makes sense to add external diodes hardwired to 5V and GND (not VIO); this maintains the 5V-tolerance on the inputs regardless of reference voltage (which is a good feature to have).

Now the question is whether we can fit this in at all. I might try something with the XBP14E5UFN, but either way, if we do this I'm pretty sure the ESD protection is going to have to go on the underside.

marcan, 2019-04-02:
Okay, I can push things around *just* enough to put the XBP14E5UFN on the top side. Very rough layout test:

[image]

electroniceel, 2019-04-02:
looking good to me. Thanks for picking this up.

whitequark, 2019-04-07:
Fixed in b9794b5.

whitequark, 2019-04-07:
Ah, not yet on SYNC.

marcan, 2019-04-07:
Fixed in 59c960b.

closed 2019-04-07 by marcan