GlasgowEmbedded/glasgow #119: Print IO bank pin mappings on silkscreen

electroniceel, 2019-04-02:
As written by me in https://github.com/GlasgowEmbedded/Glasgow/issues/68#issuecomment-478746138:
I suggest to add some silkscreen markers (for example on the back side) indicating the unconnected pins on the port headers. Not that someone expects ground there. Maybe extend that to printing the numbers, sense, Vio. I tend to forget the pinout and hate having to look it up all the time.

I don't want to put work on you with more and more of my ideas, so do you want me to provide patches? Don't want to create merge conflicts tough, they aren't as easy to see & fix in Kicad as in code.

whitequark, 2019-04-02:
I really like this idea.

marcan, 2019-04-03:
Yeah, we had to get rid of the (very vague) front side marking due to space constraints, but we can probably put some version of a full pinout on the back.

closed 2019-04-05 by electroniceel