GlasgowEmbedded/glasgow #130: Change regulator from MIC5355-S4YMME

whitequark, 2019-04-16:
This regulator appears to be out of stock or soon out of stock at multiple distributors, for unknown reasons.

electroniceel, 2019-04-17:
@marcan 
When you are re-working the layout to fit the new regulators, could you please also reposition R6 next to the regulator region: one pad of it is quite near to the mounting hole so that I fear a washer, standoff or similar hardware might short or damage the resistor.

A standard washer for M3 has 7mm outer diameter. I suggest 1mm tolerance and safety margin around it and then you end up with 9mm diameter or 4.5mm radius.

whitequark, 2019-04-17:
We should put that keepout zone on the PCB.

electroniceel, 2019-04-17:
There is the Courtyard layer, I think it is designed for such stuff

electroniceel, 2019-04-22:
The courtyard & resistors are done with #132.

marcan, 2019-04-24:
New regulators in f3fa13a

closed 2019-04-24 by marcan