GlasgowEmbedded/glasgow #137: Reduce resistor part count

electroniceel, 2019-05-01:
Reduce the number of different resistors by replacing them with similar values. Less different resistor values reduces the count of reels needed for production, reducing costs.

Candidates:
R45 (33 ohms) to 47 ohms
or
R38, R39, R43 (47 ohms) to 33 ohms
These are used for the sync io and aux.

R3 (1 Meg) to 100k
This is decoupling of the usb shield. 1 Meg is a common value for that, but 100k should not make much of a difference.

whitequark, 2019-05-01:
R38 needs to be 47 ohm to stay under AMR, so it is better to change R45.

Changing R3 to 100k is OK.

electroniceel, 2019-05-01:
heartless cost cutters might also reduce the color variance of the leds. But I think that bit of indulgence is justified.

whitequark, 2019-05-01:
Key part of the device! :D

whitequark, 2019-05-01:
(You could cut the costs by making the brightness not matched as well before that, I think.)

electroniceel, 2019-05-01:
the 6k8 R21 is just used for white. that could be changed to 4k7 or 10k

whitequark, 2019-05-01:
I'd rather keep that one.

electroniceel, 2019-05-01:
The other led resistor values seem to be used in some other circuits as well.

marcan, 2019-05-01:
Yeah, it makes more sense to choose nicely matched resistors for the LEDs and then reuse those values where they fit (which is what I did a few times).

closed 2019-05-01 by electroniceel