GlasgowEmbedded/glasgow #142: Present revC1 on http://kicad-pcb.org

herostrat, 2019-05-16:
Hey guys,

I wanted to thank you for this awesome project.
Just for the case you didn't know, there is a section on the official KiCad site were interesting projects are presented.
See: http://kicad-pcb.org/made-with-kicad/

I think this project is very interesting and would deserve a place on the list (although I am not qualified to make a decision on this). 
You can make a PR to include your board here: https://github.com/KiCad/kicad-website/tree/master/content/made-with-kicad

attie, 2023-02-23:
This repo has since moved - https://gitlab.com/kicad/services/kicad-website/-/tree/master/content/made-with-kicad

Details on submitting a project are here: https://www.kicad.org/made-with-kicad/submit-a-project/

I think this would be a good thing to do! (probably starting with the CS hardware)

whitequark, 2023-02-23:
@attie SGTM

attie, 2023-02-23:
I'm planning to submit a generic "_Glasgow revC_" (rather than "_revCx_", if that works for you...

They require a `projectdeveloper` field - with your permission, I'll list you (wq)?

whitequark, 2023-02-23:
Is that just one person? If so, `Catherine <whitequark@whitequark.org>`

attie, 2023-02-23:
Seems to be, yes - no email, so is just "_Catherine_" okay?

whitequark, 2023-02-23:
`Catherine 'whitequark'` maybe?

whitequark, 2023-02-23:
@attie SGTM

attie, 2023-02-23:
PR here: https://gitlab.com/kicad/services/kicad-website/-/merge_requests/594

herostrat, 2023-02-23:
Thanks and awesome work

closed 2023-02-23 by herostrat