GlasgowEmbedded/glasgow #147: Refactor spi-master to use SERDES + sideband design

whitequark, 2019-08-03:
This will significantly simplify a lot of downstream code.

attie, 2020-11-09:
NB: `spi-master` is now `spi-controller`, see f7aeb64.

whitequark, 2025-07-06:
No longer required (or desired), since you can rely on interface composition (i.e. instantiate an `SPIControllerInterface` and one or more `GPIOInterface`s) to achieve the same result.

This approach does require use of `await spi_iface.synchronize()` calls, which are a tripping hazard, but I think the benefits of using composition everywhere instead of copy/pasting the same GPIO sideband module into every serial interface implementation outweighs the drawbacks of the hazard.

In the future, once we have #354 (or generally, any COBS-based interface multiplexing) the synchronization calls will become unnecessary since all operations within a single applet will be inherently serializing.

closed 2025-07-06 by whitequark