GlasgowEmbedded/glasgow #151: False positive results in selftest

electroniceel, 2019-08-19:
Selftest can give false-positive results on revC, see for example https://twitter.com/szeloof/status/1161346534098833408

The main reason is probably that cmos inputs are left floating in high-z and their value is used for the test results. This seems to be a leftover from revA and revB which had hold circuits.

So selftest for revC has to be reworked to utilize the pullups/pulldowns.

For the pins-int test the internal pullups of the ice40 should be used. This is not yet supported in nmigen.

whitequark, 2019-08-19:
One problem is that there could actually be shorts on the side of pull drivers too.

electroniceel, 2019-08-19:
There could be shorts on the side of the pull drivers, there could also be a bridge over the 10k resistor, changing pullup/pulldown to push-pull.

Some of this can be detected on the other port utilizing the pins-loop and hoping that there is not the same error on the same pins.

But in the end you can detect all shorts with current measurement. This is why I wanted to have it on the test jig pcb.

esden, 2020-08-25:
Just for added information. I just ran into this issue too.

The `pins-int` selftest consistently results in:
```
E: g.applet.internal.selftest: self-test: FAIL
E: g.applet.internal.selftest: pins-int: fail high: A0 A1 A2 A3 A4 A5 A6 A7 B0 B1 B2 B3 B4 B5 B6 B7
```

All other selftests pass. This is consistent on 3 revC1 boards I could get tested.

FFY00, 2020-09-17:
Also ran into this in a new board, I had tested before with an older version and it passed.

brainstorm, 2020-10-04:
hand-assembled revC1, same here:

```
$ glasgow run selftest
I: g.device.hardware: building bitstream ID 5f3786f6a206364144276824b8f39a8e
Warning: Wire top.$flatten\U$$0.$verilog_initial_trigger has an unprocessed 'init' attribute.
Warning: Wire top.$flatten\i2c_target.\bus.$verilog_initial_trigger has an unprocessed 'init' attribute.
Warning: Wire top.$verilog_initial_trigger has an unprocessed 'init' attribute.
I: g.cli: running handler for applet 'selftest'
I: g.applet.internal.selftest: running self-test mode pins-int
I: g.applet.internal.selftest: running self-test mode loopback
E: g.applet.internal.selftest: self-test: FAIL
E: g.applet.internal.selftest: pins-int: fail high: A0 A1 A2 A3 A4 A5 A6 A7 B0 B1 B2 B3 B4 B5 B6 B7
```

whitequark, 2020-10-04:
Yes, the selftest is completely broken on revC, please stop reiterating this over and over again. I should probably just remove it.

attie, 2020-11-04:
@whitequark can this be closed after 13a3509?

whitequark, 2020-11-04:
No, we still have to implement some form of selftest on revC. I've renamed the issue.

TomKeddie, 2021-02-16:
@whitequark would you like me to look at this?  I think the first steps would be to get the existing tests to be sane on revC?

rwhitby, 2021-02-16:
@TomKeddie I'd be happy to help you with that if desired.

TomKeddie, 2021-02-16:
@rwhitby I'll look at this toward the end of the week.  I have some fixes already on a branch, I made some changes/hacks to find the shorts in the boards I built.

TomKeddie, 2021-02-25:
I have a proposed fix on https://github.com/TomKeddie/glasgow/tree/tomk_20210225_self_test am looking for people with revA, revB, revC0 and revC2 to test please.

RoganDawes, 2025-01-05:
I just started trying to use my CrowdSupply Glasgow C3, and am running into odd behaviour on the selftests. Even the pins-loop test fails, surprisingly, with different pins reported each run.

Not sure how to proceed, but it does feel like more/working selftests would help to diagnose the problem?