GlasgowEmbedded/glasgow #152: Add selftest for sync and aux pins

electroniceel, 2019-08-19:
Currently there is no selftest for the sync and aux pins of revC. These should be added.

For a loop test they should be connected like on the test jig pcb: sync to A0, Aux A10 to A1 and Aux C9 to A2.

These tests also require the internal pullups on the ice40 which are not yet supported in nmigen.

whitequark, 2019-08-19:
Sync to A0?

electroniceel, 2019-08-19:
Yeah. When executing "pins-sync-loop" the user is required to connect the sync pin to port A pin 0.

I don't like looping Aux together as they could probably have a short between them. Connecting them to port A, which was tested independently by the other tests, seems to be a better idea to me.

whitequark, 2019-08-19:
Makes sense