GlasgowEmbedded/glasgow #162: CLI handles USB device permission errors badly

emilazy, 2019-09-28:
If you run Glasgow as a user that can't access the USB device under Linux, you get a `usb1.USBErrorAccess: LIBUSB_ERROR_ACCESS [-3]` backtrace; it should print something more useful about running it with `sudo` or installing the udev rules. (Relatedly, Glasgow should probably drop permissions after opening the USB device when run as `root`.)

whitequark, 2019-09-28:
> Relatedly, Glasgow should probably drop permissions after opening the USB device when run as `root`.

I'm not sure if this works. Suppose you run it in a Docker container--you have to be root then, I think, or you can't write anywhere in it.

emilazy, 2019-09-28:
Well, it should at least drop its privileges after opening everything it needs to. This is good practice even for daemons that need to have access to root-permissioned files. However, it's hard to implement, and given that the CLI just plain crashes and dies if you try to run it in anything that isn't a proper pty with width and height it can determine, I'm not sure `glasgow.cli`-in-Docker-with-pid-namespaces is the most convincing use-case.

whitequark, 2019-09-28:
> Well, it should at least drop its privileges after opening everything it needs to.

You can't do that. An applet can decide to write a file right before termination, and it might not even know the name of this file beforehand (e.g. you're reading a floppy into `track-1.raw`, `track-2.raw` etc).

whitequark, 2019-09-28:
Also, it works just fine for me in Docker? I actually test the installation instructions that way.

whitequark, 2023-10-21:
The error that you get now (since commit 26ba5e70d12b508ebf6f95d4e8544964d28a9d05) is this:

```
$ glasgow list
E: g.device.hardware: missing permissions to open device 001/113
```

The documentation includes instructions for installing the udev rules file that should work, as far as I know, for any remotely modern Linux distribution (including eudev-based ones), so I consider this finished.

closed 2023-10-21 by whitequark