GlasgowEmbedded/glasgow #163: config/99-glasgow.rules should maybe set ENV{ID_MAKER_TOOL}

emilazy, 2019-09-29:
...or something of the sort, rather than doing group tests and assigning uaccess directly. udev ships with rules like these:

```
# software-defined radio communication devices
ENV{ID_SOFTWARE_RADIO}=="?*", TAG+="uaccess"

# 3D printers, CNC machines, laser cutters, 3D scanners, etc.
ENV{ID_MAKER_TOOL}=="?*", TAG+="uaccess"
```

electroniceel, 2019-09-29:
hmm, I'm not so sure if Glasgow fits their idea of a "maker tool". The examples they give hint more in the direction of making physical stuff than being used for electronics debugging and rev engineering.

Also this would tighten the dependencies for Glasgow: this ID_MAKER_TOOL was introduced with systemd 230, released 2016-05-21.

I'm not sure we gain anything by declaring being a maker tool. systemd will just give us the same uaccess. I actually like the GROUP="plugdev", TAG+="uaccess" approach, as it gives maximum compatibility with older and newer systems, be they systemd-based or something else.

marcan, 2019-09-30:
Just as a datapoint, eudev does not ship with such rules as far as I can tell.

whitequark, 2019-09-30:
> I'm not sure we gain anything by declaring being a maker tool. systemd will just give us the same uaccess. I actually like the GROUP="plugdev", TAG+="uaccess" approach, as it gives maximum compatibility with older and newer systems, be they systemd-based or something else.

I agree with this. I also don't find "maker tool" a convincing classification.

closed 2019-09-30 by whitequark