GlasgowEmbedded/glasgow #164: FX2_MEM is too small, needs to use 24M02 instead of 24M01

whitequark, 2019-10-02:
HX8K bitstream is about 32 Kbit too large for the 1 Mbit device, unfortunately.

I think this (+ possible DFM issues with ESD diodes?) pushes us to revC2.

electroniceel, 2019-10-02:
just changing to a 24M02 will not be enough as this will create an address clash with the fx2 eeprom.
See irc discussion for details and ideas to fix it.

whitequark, 2019-10-02:
I looked into this further, let me recap.
  1. The FX2 can boot from `1010000` and `1010001`. My understanding is that it first tries to boot from `1010000` and treats it as a 1 address byte EEPROM, and then from `1010001` and treats it as a 2 address byte EEPROM. However, this needs to be verified on actual hardware.
  2. Currently, we have FX2_MEM at `1010001` and ICE_MEM at `1010010`/`1010011`. We also have ADCs at `1010100` and `1010101`.
  3. Using the E2 strap, 24M02 can take either addresses `1010000`..`1010011` (which will not work because FX2 will interpret `1010000` as a 1 address byte EEPROM and likely fail to boot from it), or `1010100`..`1010111` (which clashes with ADCs).
  4. The ADCs can be easily remapped to `1011000`/`1011001` by changing the ADR1 strap to V_A. This does not clash with the 24M02-xR variant, which has an identification page at `1011100` when E2 is strapped high.
  5. Strapping ADCs with ADR1 to V_A deprives us of the ability to use four ADCs on the main I2C bus on revD.
  6. When the FX2 is booting from an EEPROM, it can never directly load more than 16 KB since it only has that much RAM; the largest FX2 package that exposes the external memory bus lets it have up to 64 KB of memory, but we don't use that package, so we have a 16 KB hard limit. I have used a 32 KB EEPROM just in case it will come in handy.
  7. CAT24M01 costs 0.78€@100. CAT24C256 costs ~0.18€@100. M24M02 costs 1.98€@100.

My conclusion is that, even if consolidating ICE and FX2 flashes into a single package would work (i.e. the FX2 can actually boot from a M24M02 strapped with E2 to low), it would be more expensive. I think the simplest solution is to remap accesses to ICE flash above certain addresses back to the top addresses of FX2 flash.

commit 1f3852e3 (2019-10-02): firmware: let bitstream overflow into FX2_MEM.

The HX8K bitstream is very slightly larger than ICE_MEM. It is not
practical to use a larger memory due to I2C addressing issues. Due
to my foresight of issues like this, I made FX2_MEM twice as large
as the largest FX2 firmware, so now we can stuff the ~0x1000 extra
bytes of the bitstream into FX2_MEM, and still have 0x3000 bytes
of Free Real Estate left.

This commit also reports failures during loading of flashed bitstream
by lighting up the error LED instead of ignoring them.

Fixes #164.
closed 2019-10-02 by whitequark