GlasgowEmbedded/glasgow #165: Replace ADC with INA233

electroniceel, 2019-10-03:
Replace ADC with INA233 in revD.
http://www.ti.com/product/INA233

It can monitor voltages and trigger an interrupt in cases of overvoltage, undervoltage and so on. The interrupt can then directly shut off the LDO, without interaction of the fx2 or other processor.

Also it can measure current, allowing to program a precise overcurrent limit.
Use the 0.68 Ohm resistor we have after the LDO as current shunt.

VBUS voltage input would be connected to the Vsense of the port. This allows the same flexibility as revC, but with much enhanced voltage range on Vsense, as the INA233 allows up to 36V.

whitequark, 2019-10-03:
Of note is that the ADC we are currently using can also shut off the LDO directly, in principle, but the board is not designed to take advantage of that. I am not sure if the tradeoff of somewhat worse UI and somewhat better theoretical reliability is worth it.

I think changing the FX2 code to shut off all regulators indiscriminately in case an ALERT# coincides with I2C hangup would be an acceptable fail-safe, but if there is a good way to make INA233 shut off the LDO directly that would not degrade the UI too much, we should do that.

electroniceel, 2019-10-03:
The ALERT# can directly pull down the enable signal of the LDO for the respective port. Just one extra resistor required. The FX2 can read if a over* event occured via the STATUS_* registers of the INA233.

whitequark, 2019-10-03:
How does it pull down both that and ALERT# of the FX2? I don't think polling I2C is acceptable.

electroniceel, 2019-10-03:
We could do that with diodes:
1. between ALERT# of the FX2 and the ALERT# out of the INA233
2. between LDO enable and the ALERT# out of the INA233

whitequark, 2019-10-03:
Will the added diode drop work well with the rest of the circuit? This has occurred to me, but it's not immediately obvious if there are bad implications.

electroniceel, 2019-10-03:
I suggest to use BAT54CW / BAT54-05W common cathode dual diodes in SOT-323. They are small and the max forward drop is about 300mV if we stay <1mA. I don't think this is gonna have negative effects on the function as we have CMOS logic inputs on the LDO enable and FX2 ALERT#.

The Vio LED is tied to the LDO enable. We'd have to keep that before the dropper resistor, so the LED doesn't automatically shut off when the LDO enable is pulled down by the Alert signal of the INA233. The FX2 has to switch the LDO enable off for the LED to go off. The time delay won't be visible. It is just in case the FX2 should ever lock up that the Vio LED isn't accurate. But I'd say mission accomplished if the DUT survives even in case of a lockup.

whitequark, 2019-10-03:
I think that's a reasonable thing to consider, with an assembly option that disables it in case we get it wrong.

electroniceel, 2020-07-12:
The original idea in #165 and #135 was to reuse the 0.47 Ohms dropper resistor
    we already have for the regulator as current shunt too. As the INA233 has a fixed
    measurement range of +- 81.9 mV, this would allow to measure 174 mA. My initial
    thought was that this is enough, as the specced load of the TPS73101 is 150 mA.
    
    Review and experimentation has shown that the TPS73101 can be used to 350 mA in
    most configurations before limiting set in, for a programmed output of 5 V I
    could use it even up to 420 mA. As one of the design goals of using the INA233
    is overcurrent protection, the settable range should be higher than what the
    regulator can output in best case.
    
    So the design is changed to a 0.15 Ohms current shunt and an additional 0.33 Ohms
    dropper resistor in series. This gives a total 0.48 Ohms drop for the regulator
    and a measurable current range to 546 mA.
    
    For both resistors a DNPed parallel footprint is provided as before, allowing
    the user to reduce the voltage drop if desired.

whitequark, 2020-07-12:
@electroniceel Excellent investigation!

whitequark, 2020-12-27:
This has now been done in revC2 and merged.

closed 2020-12-27 by whitequark