GlasgowEmbedded/glasgow #170: Nasty Schematic vs PCB bug in revC1 - 1.2V and 3.3V regulators are swapped

TomKeddie, 2019-10-29:
I really want this to be some kind of mistake I've made but I'm fairly sure the pcb does not match the schematic.  I believe the part number for U8 and U36 are swapped in the PCB file, the 1.2V regulator (TLV73312PQDRVRQ1) is shown as U8 driving the 3.3V bus.

This also affects the bom generated by the interactive bom generator.  I'm hoping my board is ok but once I saw the 1.2V rail rise to 1.4V and shutdown I had a sinking feeling.

You can also see this in the text of the kicad files (thank you kicad for using plain text).  The [schematic](https://github.com/GlasgowEmbedded/glasgow/blob/master/hardware/boards/glasgow/glasgow.sch#L3042) does not match the [pcb](https://github.com/GlasgowEmbedded/glasgow/blob/master/hardware/boards/glasgow/glasgow.kicad_pcb#L3341) (github won't view the file).

[image]

marcan, 2019-10-29:
Ugh. Yeah, we found this, and [fixed it](https://github.com/GlasgowEmbedded/glasgow/commits/revC1), but evidently I forgot about regenerating the fabrication outputs (and updating the PCB) with the fixed schematic data.

I'll do that as soon as I get home. Sorry that you had to stumble upon this the hard way.

TomKeddie, 2019-10-29:
Thanks I figured someone must have seen this I guess I should have been working off the branch not master.

marcan, 2019-10-29:
Those commits should be in master too, unless I did something horribly wrong. What I just didn't do yet is propagate the change to the other files.

TomKeddie, 2019-10-29:
I test generated a PR from revC1 to master, the kicad_pcb file changes were the only ones that came up not merged.

TomKeddie, 2019-10-29:
I should add that I have absolutely no ill feelings about this.  I'm in the wild west of hardware and here be dragons.  I built two boards things that the process might not be smooth, I now need to find the shorts on my other board.

marcan, 2019-10-30:
Should be fixed in dc3e083 (branch) and b436530 (master). Sorry again for leaving this pending so long :(

closed 2019-10-30 by marcan