GlasgowEmbedded/glasgow #172: revC1 schematic : lvds compatible connector part number is incorrect

TomKeddie, 2019-11-04:
The part number 20021321-00040T4LF in the schematic refers to a 40 pin connector, a 44 pin connector is required.  Unfortunately I could not find an economic 44 pin connector to recommend.

electroniceel, 2019-11-04:
For J5 itself, @esden has used Samtec FTSH-122-01-F-DV-K-P while @gregdavill has used FTS-122-01-L-DV.

According to the mating connector search at Samtec, CLP-122-02-F-D and FLE-122-01-G-DV should be possible mating connectors. Both seem to be available in single quantity at Mouser.

But I guess we should try both and check if the mechanics work out, meaning they stay on, but can still be removed without too much stress on the solder joints of J5.

electroniceel, 2020-03-03:
I now tested Samtec FLE-122-01-G-DV-A and CLP-122-02-F-D-A on the FTS-122-01-L-DV of my Glasgow:

- Both fit
- Both are availably in small quantities from Mouser
- The CLP one is more difficult to insert and remove, Samtec probably increased the tension due to the smaller height
- The FLE one looks and feels more sturdy to me

-> I would recommend the Samtec FLE-122-01-G-DV-A in the schematics

PR #188 (2020-03-03): revC schematics: fix recommended LVDS connector
Fixes #172
commit 4e0e4e33 (2020-03-03): revC schematics: fix recommended LVDS connector

Fixes #172
closed 2020-03-03 by whitequark