GlasgowEmbedded/glasgow #176: Wrong vendor ID when intitially flashing

Jan--Henrik, 2020-01-07:
Hey, if you are flashing the glasgow initially the VID:PID are 04b4:8613 (CY7C68013 EZ-USB FX2) and not 20b7:9db1, getting started with this is not intuitive. It would be nice if the initial VID:PID are also included in the udev rules.

```

SUBSYSTEM=="usb", ATTRS{idVendor}=="04b4", ATTRS{idProduct}=="8613", \
  MODE="0660", GROUP="plugdev", TAG+="uaccess"
```

whitequark, 2020-01-09:
It's not "wrong" per se. The issue here is that 04b4:8613 are programmed into the mask ROM of the FX2, and so it won't be just Glasgow that uses that pair. I don't think our udev rules should override any shared VID:PIDs by default.

Jan--Henrik, 2020-01-13:
Yea, ok, agreed

closed 2020-01-13 by Jan--Henrik