GlasgowEmbedded/glasgow #180: Multiple Simultaneous Applets

gregdavill, 2020-02-03:
For a hardware project I'm working on now I'm using both the Glasgow as a UART pty and a separate basic FTDI breakout for JTAG.

I see potential for having both a JTAG and UART interface when debugging a target. 
Or having two UARTs one for a CLI and another for Embedded Logic analyses in SoC designs.
From a hardware perspective there isn't any reason why Glasgow couldn't theoretically handle this.

But I'm not sure what that would look like from the software side, and the command line side.

I understand this feature may be impractical to support in a general sense, but do you think it may be possible for some subset of cases? Like Applet.* + UART?

whitequark, 2020-02-03:
> I understand this feature may be impractical to support in a general sense, but do you think it may be possible for some subset of cases?

I think it is, in principle, practical to support in general, and in fact, I have aimed to do so from the very beginning. It's been one of the design goals!

> But I'm not sure what that would look like from the software side,

This one is easy: you simply build/run multiple applets. I've never actually tried it, but none of the applet APIs assume that there is only one applet; there can be any amount of submodules in the target, and asyncio takes care of concurrency inherent to multiple applets.

> and the command line side.

Here... I'm not sure either, which is precisely why it isn't yet implemented.

Let's brainstorm CLI ideas in this issue?

gregdavill, 2020-02-03:
Brainstorming sounds like a good idea.

One idea, which I think could work in practice, but may be not very elegant.
The initial command which creates the multiple applets, also creates a server, each applet gets a socket or pipe interface.

In separate terminals each applet can run in a "client" mode which connects to the server to interact with the hardware. Kind of like how litex_server operates.

whitequark, 2020-02-03:
One problem is that currently applet code assumes that build/run state is one and same, and that's pretty well baked in. So these "clients" would have to be thin clients, perhaps reduced to passing the terminal fds and cwd through that socket.

But I think I could do that.

whitequark, 2020-02-03:
That still leaves open a question of how to specify multiple applets on the CLI invocation of the server.

gregdavill, 2020-02-03:
You could just specify run multiple times?
```
glasgow run uart --pin-rx 1 --pin-tx 0 -V 3.3 -b 115200 pty run uart --pin-rx 2 --pin-tx 3 -b 115200 pty
```

whitequark, 2020-02-03:
I think you can't do that with argparse. Imagine what would happen if `pty` took a filename.

gregdavill, 2020-02-04:
Turn each command into a string?

```
glasgow run "uart --pin-rx 1 --pin-tx 0 -V 3.3 -b 115200 pty" "uart --pin-rx 2 --pin-tx 3 -b 115200 pty"
```

So argparse just sees two strings? This gets ugly when filenames are required and strings need to be escaped.

Or adding a new command:
```
glasgow run-multiple
```

That drops you into a REPL or other interactive console that enabled you to add interfaces then run. If this is the path this feature heads down, it would be possible to show recently used interface combinations.

whitequark, 2020-02-04:
I think a guard argument can be done:

    glasgow run-multiple uart --pin-rx 1 --pin-tx 0 -V 3.3 -b 115200 pty -- uart --pin-rx 2 --pin-tx 3 -b 115200 pty

Still kind of ugly though.

alexhude, 2020-03-12:
+1
**run-multiple** with **--** guard argument is pretty straight forward, I vote for that :)

andresmanelli, 2020-03-13:
Hello, what about a config file? I see it like a "glasgow compose" file

`glasgow run -f config.?`

whitequark, 2020-03-13:
> what about a config file?

I don't know what a good syntax would be, and worse, currently the frontend is very closely tied to argparse.

RX14, 2020-03-14:
Would it be possible to have the interface be to run glasgow multiple times in multiple terminals and have that be handled gracefully? Then the server/daemon would be configured over some form of IPC?

whitequark, 2020-03-14:
> Would it be possible to have the interface be to run glasgow multiple times in multiple terminals and have that be handled gracefully?

Not really, since there's no partial reconfiguration possible on iCE40.

alexhude, 2020-03-14:
I think config file should be an alternative to command arguments, not a replacement. Therefore run-multiple with guard or something similar should still exist.

ChuckM, 2023-05-09:
Random thought on this somewhat dated thread ... 

One might consider both a config file and arguments. In many ways glasgow appears to be like a "hardware interpreter" in the same sense the python is a software interpreter. A glasgow "program" would consist of the applet(s) to run. It would specify at its entry point what command line arguments it would accept and how it would use them. 

I'm going to have to noodle on some syntax ideas but basically think shell script where instead of shell commands you have applets.

whitequark, 2023-05-09:
> I'm going to have to noodle on some syntax ideas but basically think shell script where instead of shell commands you have applets.

We do not need new syntax. We have perfectly good Python right here. We even have a shell-style interpreter! (`glasgow repl`)

fabianfreyer, 2023-05-09:
> We do not need new syntax. We have perfectly good Python right here. We even have a shell-style interpreter! (`glasgow repl`)

@whitequark does this mean that the ability to run multiple applets via the CLI is no longer a goal? Above, you mentioned that this is blocked on the command-line side because it's not clear how this should look. There seem to be two, not necessarily mutually exclusive, approaches:
* Shoehorn this into a command-line (`run-multiple`, etc.), which seems difficult, if not impossible using argparse.
* Specify what applets to run in some config file. This was blocked earlier by the question about a good syntax (https://github.com/GlasgowEmbedded/glasgow/issues/180#issuecomment-598722258). Does this mean this is no longer an issue, and people should just use the `script` / `real` subcommand instead? In this case, it's not really clear to me right now how one would do that easily, without e.g. mocking argparse and instantiating each frontend, which seems cumbersome.

whitequark, 2023-05-09:
@fabianfreyer Sorry, I was unclear earlier. This thread is so old I nearly forgot what I was thinking about back then...

> * Shoehorn this into a command-line (`run-multiple`, etc.), which seems difficult, if not impossible using argparse.

You could do this with argparse--it is a bit hacky though. The problem is sometimes more with applets themselves that try to do things like read from the console, or take exclusive use of limited FPGA resources. The UI is also very awkward.

People will probably want this no matter what.

> * Does this mean this is no longer an issue, and people should just use the `script` / `real` subcommand instead?

We still need a configuration file, but the surface syntax of the configuration file should just be Python.

The deep syntax, i.e. the meaning of the objects and functions used within, still needs to be defined.

Ultimately the current argparse syntax should be reduced to a special case of whatever we use for the configuration file. For that however the applets need to use some sort of language (again, Python-based) to declare the options they take.

fabianfreyer, 2023-12-25:
FWIW, here's my quick-and-very-dirty script I just used to do things, if someone wants to clean that up somehow, I could imagine it to be somewhat useful.
https://gist.github.com/fabianfreyer/b6371761fa521d14b48c602e738006b9

alexhude, 2023-12-26:
Thanks heaps! I hope I will be able to use that as reference to get JTAG+UART applet combo :)

whitequark, 2023-12-26:
@fabianfreyer This is essentially the best way to achieve what you want right now. Good job!

alexhude, 2024-04-08:
OK, as I understand maximum amount of applets is 2, because otherwise there is an error `cannot claim pipe: out of pipes`?

whitequark, 2024-04-08:
For now yes, though there is an open RFC to change that.

urjaman, 2024-07-29:
Thank you a bunch for that dual UART example - I updated and modified it a bit and made an UART (on the tty) + spi-flashrom (TCP) dual "applet". (Oh btw, it called uart0_parser twice to provide the args for uart0 and uart1 :P)

https://gist.github.com/urjaman/0e48bb2fc31bfe355cdd18a5388705a7

whitequark, 2024-07-29:
@urjaman I'd like to prototype a much better interface for achieving the same sometime soon. Would you like to participate?

urjaman, 2024-07-29:
@whitequark  I'd be happy to participate - feel free to poke me on discord or IRC when it's relevant because I don't actively "poll" github.

urjaman, 2025-03-23:
Today, in remembering how to use glasgow, i effectively did:
```
$ glasgow -h
$ glasgow run -h
$ glasgow run uart -h
$ glasgow run uart socket -h
```
(and waited 4 seconds between each)

And then I thought, maybe glasgow itself could provide a REPL/"interactive cli" (additionally to the per-applet python REPL), somewhat modeled after the CLI, so that individual operations wouldnt need to incur the python startup time.

But right after that thought, i realized that such an interface would suit itself to multiple applets, you'd "run"/configure each one, and then finally launch the whole configuration - and in this way, you could also easily store such a configuration in script form.

I know this isnt like, earth shattering insight, but I just wanted to share this to maybe provoke some more thoughts about this (is this the direction you'd like to prototype with?) - and to say that I still care about multiple applets, and also I care about having fast interactivity.

whitequark, 2025-03-23:
4 seconds is very high, what kind of device this is on? We can make it faster in various ways (and I already did my best to do so).

urjaman, 2025-03-24:
> 4 seconds is very high, what kind of device this is on? We can make it faster in various ways (and I already did my best to do so).

ASUS C201 (ARMv7, RK3288, 4 Cortex-A17 cores at 1.8Ghz, with Python 3.11.9):
```
[urjaman@veyron ~]$ time glasgow -h
usage: glasgow [-h] [-V] [-v] [-q] [-L FILE] [-F FILTER] [--statistics]
               [--serial SERIAL]
               COMMAND ...
<snipsnap tldr>
real    0m4.307s
user    0m3.723s
sys     0m0.280s
```
(okay, that was the "first run" in a while (the laptop idled all day), 3.9s now when "warmed up".)

whitequark, 2025-03-24:
That's remarkably low end. On that kind of hardware I do expect a cold run to take this long. I have an idea of how to improve this without any user-facing changes but it's a bit tricky; it involves having a long-running process that has the Glasgow modules loaded fork itself and run a `main` whenever commanded; JRuby does this (it's called "shotgun" launcher) with great efficiency.

closed 2025-05-20 by whitequark