GlasgowEmbedded/glasgow #184: Glasgow isn't quite compatible with Chrome OS

TiltMeSenpai, 2020-02-14:
Probably a low priority issue, I'm opening this mostly so I can track and share information.

Chrome OS added a preview for USB support in their linux subsystem a while back, and I was curious if I could use Glasgow with this. Initially, I would get a libusb device not found error when attempting to connect (working theory here) due to how the Glasgow board briefly disconnects when receiving firmware before coming back with the same VID/PID combo.

I also discovered the `glasgow flash` command while looking through source to find something to help. While this in theory does help Chrome with not losing track of the Glasgow board, there's something in the descriptors that Chrome must not like, bc `lsusb` doesn't show the board, and `usb-devices` freezes before showing output without the Glasgow board. After this happens, attaching USB devices to Crostini no longer does anything.

TiltMeSenpai, 2020-02-14:
Update: Found these entries in /var/log/messages:
```
2020-02-13T21:40:03.425453-08:00 ERR crosvm[6583]:  [vm_control/src/lib.rs:527] fail to recv command from usb control socket: failed to receive request or response: Resource temporarily unavailable (os error 11)
2020-02-13T21:40:03.426792-08:00 ERR vm_concierge[6545]: Failed crosvm call returned code 1
2020-02-13T21:40:03.426952-08:00 ERR vm_concierge[6545]: Failed to attach USB device: unexpected response: error: Input/output error (os error 5)
```
This looks like a bug on the ChromeOS side, but I'd like to keep this open for a bit to see if I can narrow down if there's something about Glasgow after loading device strings that is triggering this.

whitequark, 2023-02-23:
@TiltMeSenpai It's been a while. Does it work now?

closed 2023-07-02 by whitequark