GlasgowEmbedded/glasgow #187: applet.interface.i2c_initiator: timeout & error message when SCL stuck low

electroniceel, 2020-03-01:
how to reproduce:
- connect any port IO to GND (like A0)
- issue any command that uses this io as i2c initiator SCL (like "glasgow run i2c-initiator --port A -V 3.3 scan")
- glasgow is waiting endlessly for SCL to go high

I suggest to add a generous, but still reasonable timeout (like 1 second) for this case. Also return an explicit error message like "timeout, SCL stuck low" or similar.

I think this kind of error is not uncommon when testing or developing. Cases that come to mind are:

- wrong IO pin selected
- flaky connection
- external pullup resistors not connected or forgotten to enable the internal ones
- DUT misbehaving

It would be nice to add some similar detection for SDA too, currently you get back all zeros.

whitequark, 2020-03-05:
Yes, that's pretty common. We can add a timeout for USB requests, which is already supported by the streaming code we have, and raise an exception in `I2CInitiatorInterface`. I2C doesn't require a timeout (unfortunately, IMO) so it'll have to be optional, but having it something like 1 second by default seems generally OK.

whitequark, 2023-07-30:
This will become *far* easier wihth #354.