GlasgowEmbedded/glasgow #189: `platform` module shadows standard lib

DurandA, 2020-03-05:
The [`glasgow.platform`](https://github.com/GlasgowEmbedded/glasgow/tree/master/software/glasgow/platform) module shadows the [`platform`](https://docs.python.org/3.8/library/platform.html#module-platform) module from the standard Python library. This is problematic when importing libusb1 (`import usb1`) from an interpreter launched from the `glasgow` module:
```
AttributeError: module 'platform' has no attribute 'system'
```

May I suggest to rename the `platform` module to `platform_`?

whitequark, 2020-03-05:
Why are you running an interpreter from within the `glasgow` module? It's namespaced under `glasgow.` toplevel module specifically to prevent this sort of problem.

DurandA, 2020-03-05:
> Why are you running an interpreter from within the `glasgow` module?

There is no particular reason, I was just there when testing one of my scripts.

> It's namespaced under `glasgow.` toplevel module specifically to prevent this sort of problem.

Fair enough, I just thought that it was kind of evil to name modules after modules from standard lib.

closed 2020-03-05 by DurandA

whitequark, 2020-03-05:
> I just thought that it was kind of evil to name modules after modules from standard lib.

Evil? Future versions of Python can add new modules to standard library. Am I supposed to avoid those names, too? I didn't even realize there is a `platform` module.