GlasgowEmbedded/glasgow #202: Benchmark results never gets reported

FFY00, 2020-09-18:
```
$ glasgow run benchmark source
I: g.device.hardware: device already has bitstream ID e00c76bf7d43e9c1b138500bf7df1a1f
I: g.cli: running handler for applet 'benchmark'
I: g.applet.internal.benchmark: running benchmark mode source for 8.000 MiB
```

I've traced the issue to `iface.read` never continuing and bisected the breaking patch to a7d041f4ca5805a136acc8a99689833dedefd9d5.

FFY00, 2020-09-18:
Looks like that commit can be cleanly reverted, and after that everything does work as supposed to.

```
$ glasgow run benchmark
I: g.device.hardware: device already has bitstream ID e00c76bf7d43e9c1b138500bf7df1a1f
I: g.cli: running handler for applet 'benchmark'
I: g.applet.internal.benchmark: running benchmark mode source for 8.000 MiB
I: g.applet.internal.benchmark: mode source: 25.11 MiB/s (200.85 Mb/s)
```

FFY00, 2020-09-18:
The following patch solves the issue.

```patch
diff --git a/software/glasgow/support/task_queue.py b/software/glasgow/support/task_queue.py
index 25cda9f..cb87c0c 100644
--- a/software/glasgow/support/task_queue.py
+++ b/software/glasgow/support/task_queue.py
@@ -21,7 +21,8 @@ class TaskQueue:

     def _callback(self, future):
         self._live.remove(future)
-        self._done.append(future)
+        if not future.cancelled():
+            self._done.append(future)

     def submit(self, coro):
         """
```

I miss the background behind this change, @whitequark could you clarify?

whitequark, 2020-09-21:
The background for the change is described in the commit message. Whether your patch is correct or not, I can't say offhand because asyncio cancellation semantics is annoyingly tricky.

commit 1dbfc8ea (2020-10-16): support.task_queue: clear done queue after cancellation.

Since .cancel() waits until all live tasks are done, it is likely
that the done queue will contain some cancelled tasks. If the done
queue is not cleared, then the next poll()/wait_one()/wait_all() call
will cancel the caller, which usually silently terminates it. In any
case, semantically .cancel() discards the results of all tasks in
the queue even if they completed before they were cancelled.

This bug was introduced in commit a7d041f4.

Fixes #202.
closed 2020-10-16 by whitequark