GlasgowEmbedded/glasgow #204: Don't rely on argparse's internal API

FFY00, 2020-09-19:
The current CLI is relying on argparse internal API, this is undesirable. Everything I've seen so far could be replaced by our minimal custom implementation. Other option would be to move to a different argument parsing library, such as [click](https://click.palletsprojects.com/en/7.x/).

whitequark, 2020-09-21:
I reviewed a few argument parsing libraries and I didn't like any of them, so at some point likely I'll rewrite argparse.

DurandA, 2020-09-21:
> The current CLI is relying on argparse internal API

Out of curiosity, can you quote a few examples? My quick search was inconclusive.

FFY00, 2020-09-21:
https://github.com/GlasgowEmbedded/glasgow/blob/bfaf93312c2336b4f71df947a0162e99a02bbbb3/software/glasgow/cli.py#L95

whitequark, 2025-05-20:
I don't expect to be reworking the cursed argparse bits until/unless we migrate off argparse, it is just not worth the time spent.