GlasgowEmbedded/glasgow #219: Firmware support for INA233 on revC2

electroniceel, 2020-10-24:
revC2 replaces the ADC081C021, which was used on previous revisions, with the INA233.

The firmware needs to be adapted to communicate with the INA233 on revC2 boards.

First step is to reach feature-parity with revC1 (voltage sensing, interrupt for undervoltage and overvoltage).

Second step is to add current sensing and interrupt for overcurrent.

attie, 2023-02-23:
@electroniceel - is this now complete / can be closed (in #242), or are we waiting on current sense / overcurrent?

whitequark, 2023-07-24:
@attie I have a revC2 and it's unusable (I cannot enable any Vio), so I don't think it's complete...

attie, 2023-07-29:
revC3:

```
$ glasgow -vv voltage A 3.3
D: g.device.hardware: found revC3 device with serial C3-20230610T175317Z
T: g.device.hardware: USB: CONTROL OUT type=0x40 request=0x16 value=0x0000 index=0x0001 data=<00007c15> (submit)
T: g.device.hardware: USB: CONTROL OUT (completed)
T: g.device.hardware: USB: CONTROL IN type=0x40 request=0x12 value=0x0000 index=0x0000 length=1 (submit)
T: g.device.hardware: USB: CONTROL IN data=<00> (completed)
T: g.device.hardware: USB: CONTROL IN type=0x40 request=0x17 value=0x0000 index=0x0000 length=1 (submit)
T: g.device.hardware: USB: CONTROL IN data=<00> (completed)
T: g.device.hardware: USB: CONTROL OUT type=0x40 request=0x14 value=0x0000 index=0x0001 data=<e40c> (submit)
T: g.device.hardware: USB: CONTROL OUT (completed)
T: g.device.hardware: USB: CONTROL IN type=0x40 request=0x12 value=0x0000 index=0x0000 length=1 (submit)
T: g.device.hardware: USB: CONTROL IN data=<00> (completed)
T: g.device.hardware: USB: CONTROL OUT type=0x40 request=0x16 value=0x0000 index=0x0001 data=<9a0b2e0e> (submit)
T: g.device.hardware: USB: CONTROL OUT (completed)
T: g.device.hardware: USB: CONTROL IN type=0x40 request=0x12 value=0x0000 index=0x0000 length=1 (submit)
T: g.device.hardware: USB: CONTROL IN data=<00> (completed)
Port    Vio     Vlimit  Vsense  Monitor
T: g.device.hardware: USB: CONTROL IN type=0x40 request=0x17 value=0x0000 index=0x0000 length=1 (submit)
T: g.device.hardware: USB: CONTROL IN data=<00> (completed)
T: g.device.hardware: USB: CONTROL IN type=0x40 request=0x14 value=0x0000 index=0x0001 length=2 (submit)
T: g.device.hardware: USB: CONTROL IN data=<fa0c> (completed)
T: g.device.hardware: USB: CONTROL IN type=0x40 request=0x1a value=0x0000 index=0x0001 length=2 (submit)
T: g.device.hardware: USB: CONTROL IN data=<7c15> (completed)
T: g.device.hardware: USB: CONTROL IN type=0x40 request=0x15 value=0x0000 index=0x0001 length=2 (submit)
T: g.device.hardware: USB: CONTROL IN data=<0000> (completed)
T: g.device.hardware: USB: CONTROL IN type=0x40 request=0x16 value=0x0000 index=0x0001 length=4 (submit)
T: g.device.hardware: USB: CONTROL IN data=<9a0b2e0e> (completed)
A       3.3     5.5     0.0     3.0-3.6
```

whitequark, 2023-07-29:
Nevermind, I got tripped up by my own bad UI choices; I had a voltage limit set (years ago) and totally forgot about it.

I think this is done then.

closed 2023-07-29 by whitequark