GlasgowEmbedded/glasgow #239: On windows, first glasgow command after plug-in errors, second one works

0x53A, 2020-12-03:
If I plug my glasgow in the first command errors out:

1) plug in
2) run ``glasgow list`` (and I get a bling from windows as if a device was connected)

```
C:\Users\lr>glasgow list
I: glasgow.device.hardware: found revC1 device with API level 0 (supported API level is 1)
Traceback (most recent call last):
  File "C:\Users\lr\AppData\Local\Programs\Python\Python39\Scripts\glasgow-script.py", line 33, in <module>
    sys.exit(load_entry_point('glasgow', 'console_scripts', 'glasgow')())
  File "c:\glasgow\repo\glasgow\software\glasgow\cli.py", line 856, in main
    exit(loop.run_until_complete(_main()))
  File "c:\users\lr\appdata\local\programs\python\python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "c:\glasgow\repo\glasgow\software\glasgow\cli.py", line 835, in _main
    for serial in sorted(GlasgowHardwareDevice.enumerate_serials()):
  File "c:\glasgow\repo\glasgow\software\glasgow\device\hardware.py", line 174, in enumerate_serials
    devices = cls._enumerate_devices(usb_context)
  File "c:\glasgow\repo\glasgow\software\glasgow\device\hardware.py", line 107, in _enumerate_devices
    handle = device.open()
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 2153, in open
    mayRaiseUSBError(libusb1.libusb_open(self.device_p, byref(handle)))
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorNoDevice: LIBUSB_ERROR_NO_DEVICE [-4]
Exception ignored in: <function USBDevice.__del__ at 0x000001EE1FC8FF70>
Traceback (most recent call last):
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 1897, in __del__
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 1909, in close
OSError: exception: access violation writing 0x0000000000000024
Exception ignored in: <function USBDevice.__del__ at 0x000001EE1FC8FF70>
Traceback (most recent call last):
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 1897, in __del__
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 1909, in close
OSError: exception: access violation writing 0x0000000000000024
Exception ignored in: <function USBDevice.__del__ at 0x000001EE1FC8FF70>
Traceback (most recent call last):
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 1897, in __del__
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 1909, in close
OSError: exception: access violation writing 0x0000000000000024
Exception ignored in: <function USBDevice.__del__ at 0x000001EE1FC8FF70>
Traceback (most recent call last):
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 1897, in __del__
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 1909, in close
OSError: exception: access violation writing 0x0000000000000024
Exception ignored in: <function USBDevice.__del__ at 0x000001EE1FC8FF70>
Traceback (most recent call last):
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 1897, in __del__
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 1909, in close
OSError: exception: access violation writing 0x0000000000000024
Exception ignored in: <function USBDevice.__del__ at 0x000001EE1FC8FF70>
Traceback (most recent call last):
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 1897, in __del__
  File "c:\users\lr\appdata\local\programs\python\python39\lib\site-packages\libusb1-1.8.1-py3.9-win-amd64.egg\usb1\__init__.py", line 1909, in close
OSError: exception: access violation writing 0x0000000000000024
```

3) run ``glasgow list`` again:

```
C:\Users\lr>glasgow list
C1-20200924T010855Z
```

Notes: the usb driver has been a bit finicky and I clicked buttons in Zardig without really understanding what they do, apparently I currently have libusbk.

[image]

whitequark, 2020-12-03:
There's a timeout in the enumeration code which probably needs increasing, but it's at 5 seconds already and I'm somewhat surprised that isn't enough. If it *is* increased significantly then we should do this once libusb hotplug lands on Windows (https://github.com/libusb/libusb/issues/86).

0x53A, 2020-12-13:
After ``glasgow flash`` this seems to happen less often.
It still errors out if I execute a glasgow command immediatly after plugging it in, but if I plug it in and wait 10 seconds, it works.

Maybe ``I: glasgow.device.hardware: found revC1 device with API level 0 (supported API level is 1)`` could be extended with that information?

> [...] (supported API level is 1, __run 'glasgow flash' to update your device__)

whitequark, 2020-12-13:
I think it would be better to fix the hotplug issue, or perhaps do both.

commit b6e33d79 (2023-08-05): device.hardware: nudge the operator to reflash obsolete devices.

Fixes #239.
Closes #247.
PR #362 (2023-08-05): device.hardware: check if device is busy before loading firmware
On Linux multiple userspace applications can send control requests to a device without being aware of each other. Claim every interface to ensure mutual exclusion with any other application.

On other OSes device access is already exclusive.

Fixes #241.

Nudge the operator to reflash obsolete devices.

Fixes #239.
Closes #247.
closed 2023-08-05 by attie