GlasgowEmbedded/glasgow #252: recover flashing doesn't work when firmware or gateware is flashed

electroniceel, 2020-12-30:
Steps to reproduce
1. flash firmware or gateware onto the eeproms
2. short the recover jumper and release reset
3. ```glasgow factory --force --rev C2```

Result:
```
W: g.device.hardware: device 001/033 did not re-enumerate after firmware upload
E: g.cli: device not found
```
No new serial is flashed. The Glasgow still loads firmware and gateware after a manual reset.

I haven't investigated yet, but at least loading gateware takes longer than factory is waiting for. Could be the same for firmware, but the time difference isn't as obvious.

Maybe we need to upload a special firmware during factory that ignores everything in eeprom?

whitequark, 2020-12-30:
> Maybe we need to upload a special firmware during factory that ignores everything in eeprom?

Yes, and we can even get it from libfx2.

electroniceel, 2020-12-30:
> Yes, and we can even get it from libfx2.

One of the examples?

They don't have binaries checked in and we don't want to require sdcc.

I also think we should upload special firmware, because if the flashed firmware is completely broken (that is what recover is for), then we really don't want the fx2 to load it.

whitequark, 2020-12-30:
No, `boot-cypress.ihex` (it is built like a binary extension of the fx2 wheel). (I'm not sure actually if I'm uploading fx2 wheels, let me check that.)

whitequark, 2020-12-30:
Ok, there are binary wheels now: https://pypi.org/project/fx2/#files

whitequark, 2020-12-30:
Specifically, I'm referring to the firmware loaded when you run `fx2tool --bootloader`, which is needed for its EEPROM commands.

electroniceel, 2020-12-30:
Ok, so the solution for factory would be:
1. upload boot-cypress.ihex into the ram of the fx2
2. clear & rewrite the eeprom
3. tell the user to reset

Are the eeprom commands the same between boot-cypress.ihex and Glasgow firmware?

whitequark, 2020-12-30:
Yes, intentionally.

whitequark, 2020-12-30:
(You can also use fx2tool's EEPROM commands with Glasgow *without* replacing its firmware with the bootloader, although you can only access the first EEPROM that way.)

electroniceel, 2020-12-30:
Hmm, let's say you have bad gateware flashed. Once you load the Glasgow firmware into the ram of the fx2, it will start to load this gateware, right? So i think factory should also clear the FPGA EEPROM.

Can the boot-cypress.ihex do this or does it just have the commands for the FX2 EEPROM?

whitequark, 2020-12-30:
> Once you load the Glasgow firmware into the ram of the fx2, it will start to load this gateware, right?

No it won't if the configuration doesn't tell it to do that.

electroniceel, 2020-12-30:
Ah, ok, I haven't looked at that part yet. So the boot-cypress.ihex is fine then when we clear this config bit.

closed 2023-07-28 by whitequark