GlasgowEmbedded/glasgow #255: Schematics & Layout TODO for revC3

electroniceel, 2021-01-01:
A list of everything that should be decided and/or done for revC3 of the Glasgow hardware.

- [ ] Finish qualification tests for revC2 ( #220 #221 )
- [ ] Fix I2C address clash between SMBus Alert Response Address and the DAC of port A
- [ ] Add note about the SMBus Alert Response Address to the schematics
- [ ] Fix internal I2C testpoint silkscreen labels ( #250 )
- [ ] Review all changes added to revC2 between the version the prototypes were made from and merge of the branch
- [ ] Check all DFM issues @esden mentioned on stream while assembling the prototypes, are they already incorporated into revC2?
- [ ] Increase spacing between the mounting holes and the 3V3 flood fill on the internal power layer to prevent shorts when using serrated washers that could penetrate into the power layer
- [ ] Decide if we want to keep the port pinout marking on the top silkscreen

electroniceel, 2021-01-10:
@esden While reviewing the changes to revC2 since the prototypes I found these issues:

### 1. LVDS alignment holes
You mentioned that the LVDS connector alignment holes are quite large and fitting loose. They haven't been changed yet and are still 1.1mm.

The samples you sent me measure alignment pins of 1.02 x 0.98 mm.
The samples of Samtec FLE-122-01-G-DV-A I have measure 0.65mm tough. Samtec suggests a hole of 0.76mm 

So the alignment pins of Samtec and your parts have different sizes and I'd say the tolerances look reasonable in CAD, maybe JLCPCB added too much tolerance and drilled much larger in the prototypes?

So to me it looks ok, but please check and decide what to do.

### 2. Reset button part number
The reset button part no is currently "TC-1109DE-X-X". The Xes stand for ordering options for length and actuation force. Please review your ordering/manufacturing data and fill in the correct part nos.

electroniceel, 2021-01-10:
@esden I found another thing that I'm not sure about:

The new footprint for the resistor packs (R_Array_Convex_4x0402) has unequal spacing between the pads. The inner pads have 0.2 mm space between them, the space between the outer and the inner pads is 0.15 mm. I thought the issue was increased likelihood of shorts between the outer and inner pads. So moving the outer pads a bit more outside is good, but you also increased them to the point that the spacing is reduced. I don't know if that will really help.

I suggest to move them out a little bit more so that they get the same space of 0.2 mm.

electroniceel, 2021-01-20:
@smunaut I added a 4.7 µF cap to VIO_AUX now, because adding 1 µF would have meant adding a new BOM line, as we don't have 1 µF yet. 

Unfortunately I couldn't get it as close to the FPGA as I wanted, but I didn't want to reroute bigger parts of the LVDS data lines this late in the design.

You can check it in the wip-revC3-hardware branch. Maybe you find a better spot to fit it or a second 100 nF nearer to the balls?

smunaut, 2021-01-20:
4u7 is fine.

And location will have to do. That's a rather crowded area and the 'clkref' line splitting the line and also going across the whole reference plane for half the lvds lines is a bit of a bummer but with 6/6 rules there is only so much you can do :/

electroniceel, 2021-01-20:
> And location will have to do. That's a rather crowded area and the 'clkref' line splitting the line and also going across the whole reference plane for half the lvds lines is a bit of a bummer but with 6/6 rules there is only so much you can do :/

Yeah, the layout in that area is a bit crowded and convoluted, not optimized for EMI, but for keeping it at 4 layers. But I think it will still do fine as it is now.

Thanks for reviewing.

esden, 2021-03-26:
## Decide on LVDS connector alignment hole size

The part that we have sourced for the production of Glasgow revC3 has a smaller alignment pin than alternative models. This results in alignment wobble. The peg is rectangular with dimensions of 0.5mm x 0.7mm. The footprint recommendation by the connector manufacturer lists a hole of 1.1mm which is very large and does not result in a very good alignment. Based on reading of other manufacturer datasheets and recommendations, the rule of thumb seems to be 0.1mm larger hole than the alignment pin. Thus I would recommend to decrease the alignment hole size to 0.8mm.

esden, 2021-03-26:
## Fix reset button part number
Fixed in commit 84810727408d623672fd89c878565682f2e92de3 the full part number is `TC-1109DE-B-F`.

esden, 2021-03-26:
## Decide on resistor pack footprint spacing

The current footprint is based on the Samsung recommended footprint dimensions. 

[image]
https://www.samsungsem.com/resources/file/global/support/product_catalog/Chip_Resistor.pdf

The `RP104P` is the `0402x4` package type. The strange part about those dimensions is that the P1 distance indicates 0.55mm between the inner pad and the outer pad centers. Which results in a 0.15mm gap. The outer gap is not defined otherwise only the center gap which is 0.2mm. I guess we could move the pad out by 0.05mm so that the gaps are balanced.

On another note, looking through a bunch of datasheets the bigger outer pads recommendation is not the most common one. (if the manufacturer recommends any footprint at all) I tried to resolve it by going to JEDEC standard. But so far I was not able to locate that. Based on [one stack overflow post](https://electronics.stackexchange.com/questions/101079/what-standard-governs-the-dimensions-of-smt-chip-resistors-and-capacitors) it seems that there is no up to date standard for resistor array footprints at the moment. But it would be nice to at least find a copy of the IEC 60115-8 document.

I would for now leave the footprint as it is, as I know that it works in production. But if we can get some better industry recommendation we can consider updating it.

esden, 2021-03-31:
## Schematics+BOM: adapt LEDs and LED-resistors

Resolved with: bd0d5b1d4d3006b3629e85382b80c47547cc3e1b

esden, 2021-03-31:
## Decide on LVDS connector alignment hole size
Changed to 0.76mm diameter holes, and removed pad chamfers that are not needed any more in: ae8d47c5f9998be509b4eca82c461dd58cf31335

attie, 2023-02-23:
@esden / @electroniceel - is this still in progress?

esden, 2023-07-31:
I have also corrected the resistor pack footprint. That was the last remaining todo item. I have pushed all the remaining patches to the wip branch. I will open a PR as soon as I rebase it onto main.

whitequark, 2023-07-31:
@esden Excellent work!! So excited to have this ***Done***.

esden, 2023-07-31:
Yeah it is indeed quite a relief! I hope everything works as expected :see_no_evil:

whitequark, 2023-10-06:
Thanks everyone! We now have all the design files upstream.

PR #378 (2023-10-06): revC3 hardware
This is the newest version of the Glasgow hardware. This is what we are shipping as part of the [CrowdSupply](https://www.crowdsupply.com/1bitsquared/glasgow) campaign.

I think we are now ready to merge it to main.

Closes: #255
closed 2023-10-06 by whitequark