GlasgowEmbedded/glasgow #256: Decide on port pinout marking on the top silkscreen

electroniceel, 2021-01-01:
[image]
[image]

During revC2 development this was added, mostly as an experiment to see how it works out in the prototypes.
Do we want to keep it for future revisions or remove it in revC3?

Pros:
- We have no other marking on the top side that shows the port assignments
- Does not cost us anything

Cons:
- Hard to read
- Does not look pretty in it's current state
- The marking is on the outside of the port headers, because that is the only space available. But the outside pins are mostly GND, the numbered pins are inside. The user could get confused by this and has to memorize it.

The protoypes were made at JLCPCB, the final production boards are planned to be made at King Credie. The latter is supposed to have higher quality silkscreen, at least available as an option. So it may look better on the final boards.

There are full pin markings without any confusion issues on the backside silkscreen. Also the aluminium case will have the pinout printed on top. People not using the aluminium case can stick marking tape on the connector shrouds or the lasercut case.

I'm yet undecided, but more leaning for removal because of the inside/outside confusion issue.

electroniceel, 2021-01-01:
@esden Could you comment on the silkscreen quality expected by King Credie? Also your opinion on this issue of course.

esden, 2021-01-01:
@electroniceel Regarding the labeling. I think I agree, I am not really using those labels myself (they are as we expected too small to be useful). I also think they might result in confusion as they are on the wrong side of the connector anyways. I think we should remove them. The wire harnesses and their labeling, as well as the markings on the back of the board turn out to be much more useful in practice.

whitequark, 2021-01-01:
I agree with the removal. Parallax makes the labels easy to accidentally misuse.

electroniceel, 2021-01-01:
Ok, let's remove them

closed 2021-01-01 by electroniceel