GlasgowEmbedded/glasgow #259: revC LED type and resistor selection

esden, 2021-01-06:
In the effort to optimize pricing for revC batch production I have ended up using different LED than the ones listed in the design files. Thus the optimized resistors for even and correct brightness ended up not being correct.

I was hoping to get it close enough with some existing BOM lines but it turns out that did not work out.

This issue is meant to track progress on finalizing the LED type and inline resistor selection to meet the following requirements:

* Reasonabely even brightness between all the different color LEDs
* Not too bright to be blinding as a naked PCB
* Bright enough to be well visible through lightpipes of the case in bright video recording brightness conditions.
* Easily sourcible LEDs for a CM at good price point.

TODO: I will post the old and new LED MFN and assoctiated resistor selections a bit later.

attie, 2021-01-06:
@esden - if you're swapping the LEDs, I'd vote for the design files to be updated to reflect the actual parts used - especially if you're going to tune the resistors again.

revC2 is already "released", so don't change that... but perhaps this update can be done for revC3? (which I believe you'll be shipping for the CS campaign?)

esden, 2021-01-06:
For revC2 design files I should have not touched the resistor values as I did not change the LED MPN. When populated with the original parts and resistor values as marcan selected will be fine. My mistake here was to touch the resistor values in the design files.

As for the parts that I have selected, I am concerned that hose parts are not as widely available from the usual part suppliers. They are easy to get in china but not from western suppliers. So at the end I am not sure what the correct end solution should be. I wish KiCad supported alternate population sets...

timonsku, 2021-01-06:
I wouldn't worry too much about the light pipes in terms of brightness. The extra size doesn't really decrease brightness, it just means the light pipes have to be longer. Acrylic has 90%+ translucency so the brightness loss is minimal, making them not blindingly bright is more important imo.
Also note that a very high brightness can also cause more bleed as we are not working with "true" light-pipes that transfer the light of each LED individually but rather packed together as one big blob.

For the case I only would like to see at least a different LED color for the FX2 LED so they can be differentiated when they inevitably bleed into each other a bit with the light-pipe. The ICE green led and the orange ACT led are also a bit similar in color and probably hard to differentiate for color blind people.
Blue for ICE and white for FX2 sound like good options to me as these colors are used on the user LEDs already.

electroniceel, 2021-01-06:
> As for the parts that I have selected, I am concerned that hose parts are not as widely available from the usual part suppliers. They are easy to get in china but not from western suppliers.

If the new LEDs are available from LCSC that would be good enough for me. They reliably ship to western countries and don't ship fakes or misuse part nos.

In the end the LEDs are not a dealbreaker for rebuilding. If someone is able to solder BGAs and 0402 to build their own Glasgow, I'd say they should usually be experienced enough to change to different LEDs and matching resistors if they can't source the original ones.

But I second @attie in that we should put the parts actually used in production into the schematics and BOM.

marcan, 2021-01-07:
Yeah, the canonical schematics and BOM should just have *some* set of resistors and LEDs that are known to be properly matched, ideally the production ones.

People DIYing this will probably use whatever they have on hand, and people doing their own production runs get to redo this process. I'm not too concerned as long as the design files reflect one known-good set of parts, and it might as well be @esden's production version.

As for the brightness, I'm happy as long as long as the end result is roughly similar to what I came up with. For referecence, [this](https://twitter.com/marcan42/status/1103642165475864577) is how I did it.

Basically, I would say: naked LEDs should be discernible (not necessarily easily) in bright sunlight. Lightpipes should be easily visible in normal indoor bright lighting conditions. And the things shouldn't light up the whole room when in pitch darkness. The revC1 I got from @esden has pretty good brightness, so that's probably a good thing to match to (except the Vio LEDs are way too dim on that one; I think the resistor choice was wrong for that or something and I assume that's been fixed already).

electroniceel, 2021-01-07:
> For the case I only would like to see at least a different LED color for the FX2 LED so they can be differentiated when they inevitably bleed into each other a bit with the light-pipe. The ICE green [...]

The majority of the orders seems to be with the al. case, so we should really make the  LEDs work well for that scenario too. I think @timonsku has a valid point here, light bleed can be an issue, especially when looking at the LEDs from an angle.

Even though LEDs are surprisingly hard to photograph an still look realistic, I tried to emulate the lightpipe setup with a slightly frosted acrylic sheet I had around:

[image]

The picture above is from straight above. But when looking at an angle, color bleed and axis shift become a problem:

[image]

To see how @timonsku s proposal would look like, I took the user LEDs from my prototype and resoldered them to the status LEDs:

[image]

While I think the look on the bare board is a bit more "nervous" than the greens that were suggesting everything is ok, I still think it is ok. Also the white and blue are much brighter than the orange activity LED.

But unfortunately it doesn't work out for me through the acrylic:

[image]

The white has a bluish tint and the blue isn't visible as a point, more as a color field bleeding everywhere. I guess this mostly comes from these individual LED models, but could also have to do something with the light permeability of the acrylic being dependant on wavelength.

So unfortunately I don't have a nice solution for the problem.

esden, 2021-03-26:
Michael Ossmann tipped me off that they have put a bunch of effort into selecting LCSC sourcable LEDs with appropriate resistors for [Luna](https://github.com/greatscottgadgets/luna). For reference here is their selection of led and resistor values:
* Red, OptoSupply, OSR50603C1E, 1k
* Orange, EKINGLUX, E6C0603SEAC1UDA, 1k
* Yellow, EKINGLUX, E6C0603UYAC1UDA, 1k
* Green, Orient, ORH-G36G, 10k
* Blue, Orient, ORH-B36G, 2k2
* Purple, OptoSupply, OSVX0603C1E, 2k2 (does not seem to be available through LCSC?)
* Ice Blue, MEIHUA, MHT192WDT-ICE, 5k5 (5k5 is not a common value, 5k49 or 5k6 could be used here?)
* Pink, OptoSupply, OSK40603C1E, 2k2
* White, Orient, ORH-W46G, 2k2

The resulting color balance picture by ktemkin: https://twitter.com/ktemkin/status/1357120015133077504

esden, 2021-03-29:
I like the @electroniceel suggestion to change the `STATUS LED` colors. I have put the proposed color scheme onto a test board and it does look good and makes it easier to distinguish them. (will test on a live glasgow board next)

[image]

This is the proposed color scheme:
* PWR: Green
* FX2: White
* ICE: Blue
* ACT: Orange
* ERR: Red
* U1: Blue
* U2: Pink
* U3: White
* U4: Pink
* U5: Blue

The part numbers used on my board with resistor values:
* Green, Foshan NationStar Opto, NCD0603G1, 10k
* White, Foshan NationStar Opto, NCD0603G1, 2k2
* Blue, Orient, ORH-B36G, 2k2
* Orange, Foshan NationStar Opto, NCD0603O1, 1k
* Red, Orient, ORH-R36A, 1k
* Pink, OptoSupply, OSK40603C1E, 2k2

The brightness match seems acceptable to me. It is still possible to see that the led are on with a bright flashlight shining onto the board, and they are also not blindingly bright. (it is very hard to convey on camera though)

electroniceel, 2021-03-29:
did you check the proposed LEDs through a lightpipe?

As noted above, I had problems when looking at an angle and especially with the blue LEDs.

electroniceel, 2021-03-29:
>     * Green, Foshan NationStar Opto, NCD0603G1, 10k
>     * White, Foshan NationStar Opto, NCD0603G1, 2k2

these are the same P/N, this seems like a mistake.

I use NCD0603G1 myself, they are my default green LEDs.

esden, 2021-03-29:
After testing multiple lightpipes, I am convinced that using clear acrylic is the way to go. At least as long as we keep the oval lightpipes. Diffusion will only be useful if we have another way of separating the LED, for example by using one light pipe per LED. Having different colors of LED next to each other is still desirable though as it makes it easier to see what is going on at a glance.

I corrected the P/N above.

esden, 2021-03-31:
Made the proposed changes and committed in: bd0d5b1d4d3006b3629e85382b80c47547cc3e1b
For additional reference, I showed off the selected LEDs and discussed their visibility through the light pipes during [my twitch stream](https://www.twitch.tv/videos/969238435).

esden, 2021-03-31:
I consider the issue closed. If there are still questions or need for followup please feel free to reopen the issue.

closed 2021-03-31 by esden