GlasgowEmbedded/glasgow #263: applet.interface.uart: high Baud rates drop data

attie, 2021-01-19:
It seems that using ~1.5Mbaud, with high data rates can cause the applet FIFOs to overflow, and silently drop data.

1.5Mbaud is worst-case of ~150KB/s... this overflow is likely due to the relatively inefficient USB utilization that is incurred with the default `auto_flush=True`.

Suggestions:

1. Set `auto_flush=False`, and implement a timer to flush buffers after a number of idle bit/word-times
2. Increase FPGA-side FIFO size, past the default of `depth=512`
3. Add a register to indicate overflow / dropped frames

whitequark, 2023-07-24:
We should probably implement something like Nagle for UART.

whitequark, 2025-07-06:
Item (1) will be fixed globally in #794.
Item (2) isn't necessary once (1) and (3) are addressed.
Item (3) is fixed in #932.

commit 73a91184 (2025-07-13): gateware.fx2_crossbar: flush IN FIFO once per microframe.

This eliminates a common pitfall for applet authors, where by default
the IN FIFO is flushed too often, and very small packets clog the FX2
buffers, causing buffer overflows and data loss. With this change, so
long as the host polls the device once per microframe (with XHCI hosts,
this can happen even more than once), there will never be a case where
an FX2 buffer isn't available, unless the bus is very heavily loaded
with other traffic.

There are two cases where more control over flushing is desired.

First, in request/response style [... cut]
closed 2025-07-13 by whitequark