GlasgowEmbedded/glasgow #264: VGM protocol:

hansfbaier, 2021-01-21:
VGM file:
[Magical Sound Shower](https://vgmrips.net/packs/vgm/Arcade/SegaSys/Out_Run_%28Arcade%29/01%20Magical%20Sound%20Shower.vgz)
```
$ python3 vgm2tsv.py MagicalSoundShower.vgz MagicalSoundShower.tsv
Traceback (most recent call last):
  File "vgm2tsv.py", line 18, in <module>
    asyncio.run(reader.parse_data(player))
  File "/usr/lib/python3.8/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/jack/.local/lib/python3.8/site-packages/glasgow-0.1.dev1+gb1985b3-py3.8.egg/glasgow/protocol/vgm.py", line 188, in parse_data
    raise NotImplementedError("Unknown VGM command {:#04x} at stream offset {}"
NotImplementedError: Unknown VGM command 0x67 at stream offset 128
```

whitequark, 2021-01-21:
The gist you've been using is a tech demo. It is not supported.

closed 2021-01-21 by whitequark

hansfbaier, 2021-01-21:
To me, judging from the stacktrace it looks like the code in the gist is not the issue. But never mind, my life does not depend on it 🤗

whitequark, 2021-01-21:
Yes, and the code in this repo isn't going to change to support that gist. Like I said, it's a tech demo.