GlasgowEmbedded/glasgow #276: applet: explore ideas for testing usb 2.0 data with Glasgow

TomKeddie, 2021-02-16:
I'm opening this issue to document my experiments with Luna (https://github.com/greatscottgadgets/luna) on glasgow.

Unsurprisingly Luna works fine using the lvds connector.  Here is the platform definition I used.  I used a usb direct pmod and my lvds-pmod adapter to test this.

Next step is to test using the test interface.

```
#
# This file is part of LUNA.
#
# Copyright (c) 2020 Great Scott Gadgets <info@greatscottgadgets.com>
# SPDX-License-Identifier: BSD-3-Clause

""" TinyFPGA Platform definitions.

This is a non-core platform. To use it, you'll need to set your LUNA_PLATFORM variable:

    > export LUNA_PLATFORM="luna.gateware.platform.glasgow:GlasgowPlatform"
"""

import os
import subprocess

from nmigen import Elaboratable, ClockDomain, Module, ClockSignal, Instance, Signal, Const, ResetSignal
from nmigen.build import Resource, Subsignal, Pins, Attrs, Clock, Connector, PinsN
from nmigen_boards.resources import *
from nmigen.vendor.lattice_ice40 import *

from .core import LUNAPlatform

class GlasgowDomainGenerator(Elaboratable):
    """ Creates clock domains for the TinyFPGA Bx. """

    def __init__(self, *, clock_frequencies=None, clock_signal_name=None):
        pass

    def elaborate(self, platform):
        m = Module()
        locked = Signal()

        # Create our domains...
        m.domains.sync   = ClockDomain()
        m.domains.usb    = ClockDomain()
        m.domains.usb_io = ClockDomain()
        m.domains.fast   = ClockDomain()
        # ... create our 48 MHz IO and 12 MHz USB clock...
        # Fout = 48MHz * (47 + 1) / (2^4 * (2 + 1)) = 48MHz
        clk48  = Signal()
        clk12  = Signal()
        m.submodules.pll = Instance("SB_PLL40_2F_CORE",
            i_REFERENCECLK  = platform.request(platform.default_clk),
            i_RESETB        = Const(1),
            i_BYPASS        = Const(0),

            o_PLLOUTCOREA   = clk48,
            o_PLLOUTCOREB   = clk12,
            o_LOCK          = locked,

            # Create a 48 MHz PLL clock...
            p_FEEDBACK_PATH = "SIMPLE",
            p_PLLOUT_SELECT_PORTA = "GENCLK",
            p_PLLOUT_SELECT_PORTB = "SHIFTREG_0deg",
            p_DIVR          = 2,
            p_DIVF          = 47,
            p_DIVQ          = 4,
            p_FILTER_RANGE  = 1,
        )

        # ... and constrain them to their new frequencies.
        platform.add_clock_constraint(clk48, 48e6)
        platform.add_clock_constraint(clk12, 12e6)

        # We'll use our 48MHz clock for everything _except_ the usb domain...
        m.d.comb += [
            ClockSignal("usb")     .eq(clk12),
            ClockSignal("sync")    .eq(clk48),
            ClockSignal("usb_io")  .eq(clk48),
            ClockSignal("fast")    .eq(clk48),

            ResetSignal("usb")     .eq(~locked),
            ResetSignal("sync")    .eq(~locked),
            ResetSignal("usb_io")  .eq(~locked),
            ResetSignal("fast")    .eq(~locked)
        ]
        return m

class GlasgowPlatform(LatticeICE40Platform, LUNAPlatform):
    name                   = "Glasgow"
    clock_domain_generator = GlasgowDomainGenerator
    default_usb_connection = "usb"
    device                 = "iCE40HX8K"
    package                = "BG121"
    default_clk            = "clk48"
    resources   = [
        Resource("clk48", 0, Pins("K6", dir="i"),
                 Clock(48e6), Attrs(IO_STANDARD="SB_LVCMOS")),

        *LEDResources(pins="G9", attrs=Attrs(IO_STANDARD="SB_LVCMOS")),

        DirectUSBResource(0, d_p="H2", d_n="H1", pullup="J2", d_p_oe="C1", d_n_oe="D2", attrs=Attrs(IO_STANDARD="SB_LVCMOS33")),
    ]
```

whitequark, 2021-02-16:
To keep my workload as a maintainer manageable I would strongly prefer to avoid supporting any third party frameworks on Glasgow hardware even if it is something as nice as Luna. I will close this issue because of this.

closed 2021-02-16 by whitequark

TomKeddie, 2021-02-16:
Thanks @whitequark I was expecting this kind of thing would come to Glasgow with support for external applets, we not expecting this to be in tree.  I understand your need to stay focussed.  I have some more stuff to document that I'd like to add here (if you don't mind).  If nothing else, to aid someone in the future that might want to build usb test on Glasgow.

whitequark, 2021-02-16:
I understand that this is frustrating but unless explicitly limited, open source tends to creep in scope until the maintainers fold. I will preventively limit the scope to avoid that. I think I made the board easily hackable for anyone who wants to do that with the understanding that they are completely on their own.

TomKeddie, 2021-02-16:
Not frustrated, a good call on your part I think.

whitequark, 2021-02-16:
The LVDS bank in particular is essentially something that exists because @marcan didn't want any FPGA I/O to go to waste. It is not even going to be 100% tested (I think, this might not be final) and it will go away in future revs.

TomKeddie, 2021-02-16:
Agreed, I used lvds as a PoC for the logic and nothing else.  My next comment was to be about resolving whether this is possible for the test interface.  I'm not sure it is.

TomKeddie, 2021-02-16:
(am leaving this here for future explorers).

The real challenge for driving the usb data lines directly with the glasgow test interface will be the direction control on the level converters.  I've modified luna to wire out the direction control to pins and used them to drive the test interface (A[0:2]) with this change to the platform definition.
```
        DirectUSBResource(0, d_p="A1", d_n="A2", pullup="B3", d_p_oe="C7", d_n_oe="C8", attrs=Attrs(IO_STANDARD="SB_LVCMOS33")
```
There are signs of life but it's not clean.  The messages are corrupt are the enumeration fails.
```
[788582.485186] usb 2-4.7.4: new high-speed USB device number 79 using ehci-pci
[788582.593700] usb 2-4.7.4: New USB device found, idVendor=03fd, idProduct=000f, bcdDevice= 0.00
[788582.593704] usb 2-4.7.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
```

rwhitby, 2021-02-16:
This is all good for me to hear and understand.   I was contemplating putting D+/D- switchable pull-up/down on the Glasgow USB-PD board I'm designing (https://github.com/rwhitby/glasgow-addons/tree/usbp-pd-addon/hardware/usb-pd-addon) but this discussion is now dampening that contemplation and pushing me towards just making the USB-PD board easily connectable to a supported Luna board (Glasgow/USB-PD handles the USB-PD side, Luna handles the USB data side).