GlasgowEmbedded/glasgow #279: README updates for macOS

jspencer, 2021-02-26:
I don't have hardware yet but I thought I'd run the install and poke around. 

For macOS I had to run
`python3 setup.py develop` instead of `python setup.py develop

It's possible that stock Macs ship with python3 as the default but I installed (the OS) via migration.
It might be worth adding a note to the README as the error message is not especially helpful:

```
running develop
error: can't create or remove files in install directory

The following error occurred while trying to add or remove files in the
installation directory:

    [Errno 13] Permission denied: '/Library/Python/2.7/site-packages/test-easy-install-87242.write-test'

The installation directory you specified (via --install-dir, --prefix, or
the distutils default setting) was:

    /Library/Python/2.7/site-packages/

Perhaps your account does not have write access to this directory?  If the
installation directory is a system-owned directory, you may need to sign in
as the administrator or "root" account.  If you do not have administrative
access to this machine, you may wish to choose a different installation
directory, preferably one that is listed in your PYTHONPATH environment
variable.

For information on other options, you may wish to consult the
documentation at:

  https://setuptools.readthedocs.io/en/latest/easy_install.html

Please make the appropriate changes for your system and try again.
```

Installing on Linux Mint was fine after installing `icestorm` which was not initially obvious.

attie, 2021-02-26:
Python 2.7 isn't supported, but I agree, it might be sensible to mention `python3` instead of `python` - are you willing to make a PR for this?

The error message is indeed not very helpful - this would likely be resolved by using a virtual environment (`venv` / `virtualenv`), which I'd suggest is the preferred way to work with Glasgow, and I'd suggest that should be included in the docs too.

Re installing `icestorm`, did you see and follow this bit in the setup guide?

> You will also need Yosys and nextpnr-ice40, both from the master branch. Follow the setup instructions for [Yosys](https://github.com/yosysHQ/yosys/#setup) and [nextpnr](https://github.com/YosysHQ/nextpnr/#nextpnr-ice40).

jspencer, 2021-02-26:
OK great, thanks. I'll make a PR. My concern was that there may be cases where `python` points to Python 3.x and a `python3` alias is missing, but from reading around, most people wouldn't have their path and aliases set up that way, so it seems reasonable.

And yes, now I see the note to install IceStorm in the nextpnr README. I must have glossed over that part. Oops!

As for the virtual environment, I'm not familiar enough to write a guide but I'd be more than happy to run it though its paces on a few systems if someone adds that to the readme.

closed 2021-03-01 by jspencer