GlasgowEmbedded/glasgow #286: applet.memory-25x: Sector parameter, doesn't update erase opcode

gregdavill, 2021-11-13:
When using the memory-25x, you have the ability to set the sector size, using the `-S` parameter. 
However setting this value does not update the command that's actually sent to the flash.

By default the opcode `0x20` is sent, which (on common flashes) is a 4kB sector erase.

Most flash devices supports the 64kB sector erase, which typically executes faster, but in order to make use of this the opcode sent will need to change based on the `-S` parameter.

whitequark, 2023-10-26:
@gregdavill In case SFDP is available this is easy to handle. What is to be done if there's no SFDP data available? Should an option to specify the exact erase opcode be mandatory in this case?

gregdavill, 2023-10-26:
Using SFDP makes the most sense when it's available as a default option, it can pick the largest erase sector size based on the amount of data to write.

If an opcode is presented it should take precedence over the SFDP. (But show a WARN/INFO if SFDP is available and the opcode is not present in the SFDP)

If no SFDP is available, requiring the opcode to be provided is a good option.

whitequark, 2023-10-26:
Got it! I'll implement this soonish (and then finally take care of your other PRs.)

whitequark, 2026-09-16:
The `memory-25q` applet should have fixed this.

closed 2026-09-16 by whitequark