GlasgowEmbedded/glasgow #287: GPIB applet

FFY00, 2021-11-13:
It would be useful to have a GPIB applet, as it would allow us to talk with lots of, mainly older, testing equipment.

whitequark, 2021-11-13:
Yeah. I've considered it; I'm not sure if I have anything around that supports GPIB.

whitequark, 2021-11-13:
If you can suggest something cheap (bonus points if it's useful) I can buy then I might just do that and implement it.

FFY00, 2021-11-13:
I really need to get started on hardware design so I can help with this stuff.

Unfortunately, I don't think I know anything super cheap, but there are many cheap-ish, for what you get at least, useful legacy gear.

I really like the HP/Agilent 66xxB lines of power supplies. They are pretty good, and you can easily put binding posts on the front (there are holes already in the front panel plastic piece and the PCB is prepared, just needs some missing jumpers). I don't know about shipping/import costs for where you are, but you can find them pretty cheap in the US, a bit more expensive for the rest of the world. Here are a couple listings in Asia:

https://www.ebay.com/itm/110943616006
https://www.ebay.com/itm/153823238670
https://www.ebay.com/itm/153283612160
https://www.ebay.com/itm/272327301175
https://www.ebay.com/itm/392242863402
https://www.ebay.com/itm/392241077278
https://www.ebay.com/itm/283952534100
https://www.ebay.com/itm/113411427377
https://www.ebay.com/itm/401709583912

Some other useful stuff:

https://www.ebay.com/itm/294203055354
https://www.ebay.com/itm/133922320443

I have just bought a Tektronix TDS744A in a lot, which can be hacked to 1GHz. That is a bit more expensive, but still could be potentially interesting.

whitequark, 2021-11-13:
All of those either can't be shipped, or have >200$ shipping...

FFY00, 2021-11-17:
😕 

This is not cheap, but reasonably priced, I think, if it doesn't give you extra shipping costs.

https://www.ebay.com/itm/HP-Agilent-34401A-6-1-2-Digit-Digital-Multi-Meter-Main-Unit-Only-From-JAPAN-/384484576020?mkcid=16&mkevt=1&_trksid=p2349624.m46890.l6249&mkrid=711-127632-2357-0

whitequark, 2021-11-17:
Nope, can't be shipped.

whitequark, 2023-10-21:
I have several GPIB devices and am working on the GPIB controller applet with some others. This turns out to be unexpectedly tricky because *holy shit have you seen the spec?!?*