GlasgowEmbedded/glasgow #292: Running glasgow applets fails on first try after device plug-in

thasti, 2022-04-13:
On recent ArchLinux (Kernel 5.6.13), glasgow fails to run any applet on the first invocation of `glasgow run` after connecting the device to USB.  The error message is the following:

```
[thasti@nyan microbeam]$ glasgow run ...

Traceback (most recent call last):
  File "/home/thasti/.local/bin/glasgow", line 33, in <module>
    sys.exit(load_entry_point('glasgow', 'console_scripts', 'glasgow')())
  File "/home/thasti/projekte/glasgow/glasgow/software/glasgow/cli.py", line 857, in main
    exit(loop.run_until_complete(_main()))
  File "/usr/lib/python3.10/asyncio/base_events.py", line 641, in run_until_complete
    return future.result()
  File "/home/thasti/projekte/glasgow/glasgow/software/glasgow/cli.py", line 470, in _main
    device = GlasgowHardwareDevice(args.serial)
  File "/home/thasti/projekte/glasgow/glasgow/software/glasgow/device/hardware.py", line 179, in __init__
    devices = self._enumerate_devices(usb_context, _factory_rev)
  File "/home/thasti/projekte/glasgow/glasgow/software/glasgow/device/hardware.py", line 107, in _enumerate_devices
    handle = device.open()
  File "/home/thasti/.local/lib/python3.10/site-packages/libusb1-3.0.0-py3.10.egg/usb1/__init__.py", line 2055, in open
    mayRaiseUSBError(libusb1.libusb_open(self.device_p, byref(handle)))
  File "/home/thasti/.local/lib/python3.10/site-packages/libusb1-3.0.0-py3.10.egg/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/home/thasti/.local/lib/python3.10/site-packages/libusb1-3.0.0-py3.10.egg/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorNoDevice: LIBUSB_ERROR_NO_DEVICE [-4]
```

This error message coincides with a re-enumeration of the device in dmesg:

```
[397361.999956] usb 2-2: USB disconnect, device number 94
[397362.290842] usb 2-2: new high-speed USB device number 95 using xhci_hcd
[397362.433003] usb 2-2: New USB device found, idVendor=20b7, idProduct=9db1, bcdDevice= 1.31
[397362.433017] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[397362.433023] usb 2-2: Product: Glasgow Debug Tool
[397362.433027] usb 2-2: Manufacturer: whitequark research
[397362.433031] usb 2-2: SerialNumber: C1-20200924T000714Z
```

Any further attempts at running applets work just fine every time, however the issue can be reproduced (on multiple machines here) by disconnecting and reconnecting the device.

attie, 2022-04-13:
Where did your Glasgow come from? (self build / me?)

Please could you try running the following?

```
glasgow flash
```

thasti, 2022-04-16:
Hi Attie. This indeed is one from the batch you made. I confirm that after running `glasgow flash`, the issue disappears. Now, the glasgow already enumerates fully (i.e. as `Glasgow Debug Tool`) on the initial plug-in - not only after launching the `glasgow run` command. 

So now I'm not sure if "device is being used without firmware in flash" is a valid use case or not, and therefore whether this issue is actually one or not.

attie, 2022-04-16:
Thanks for confirming - I tried to be careful and flash all boards, so I'm surprised you needed to do that step.... my apologies either way.

This is a valid use-case, and it should load firmware and "just work" on the first run... I have a feeling there is already a ticket about this, so I'll look through later and link / close as necessary.

attie, 2022-04-16:
This seems to be similar in behaviour to / the same as #239...

whitequark, 2022-04-16:
On Linux, Glasgow uses hotplug detection since it's available in libusb, so #239 shouldn't apply...

whitequark, 2023-07-24:
To investigate this further I would need verbose logs from `glasgow -vvv` and also timestamped `dmesg`. Let me know if this is something you can still reproduce, please.

whitequark, 2023-10-06:
No response from reporter for a few months; closing.

closed 2023-10-06 by whitequark