GlasgowEmbedded/glasgow #294: Installing with Python 3.10

andresmanelli, 2022-11-15:
Hello,

I tried installing the software from Python 3.10 and I had trouble with `aiohttp`. Apparently it's resolving to version 4.0.0, but version 3.8.1 is newer (See [this comment](https://github.com/aio-libs/aiohttp/issues/6898#issuecomment-1233074699)).

My workaround was to apply this patch:

```diff
diff --git a/software/setup.py b/software/setup.py
index 289ead7..0b7e0e1 100644
--- a/software/setup.py
+++ b/software/setup.py
@@ -29,7 +29,7 @@ setup(
         "amaranth",
         "fx2>=0.9",
         "libusb1>=1.8.1",
-        "aiohttp",
+        "aiohttp==3.8.3",
         "pyvcd",
         "bitarray",
         "crcmod",
```

Maybe this is a symptom for another problem, but this worked for me.

whitequark, 2022-11-15:
Thanks, I'll take a look.

whitequark, 2023-02-07:
Thanks for the report! Should be fixed now.

commit 7d8db81b (2023-02-07): software: tighten dependency to aiohttp~=3.8.

Fixes #294.
closed 2023-02-07 by whitequark