GlasgowEmbedded/glasgow #297: The `asyncio.coroutine` decorator has been removed in Python 3.11

VioletEternity, 2022-12-17:
The `asyncio.coroutine` decorator has been [deprecated since version 3.8](https://docs.python.org/3.10/library/asyncio-task.html?highlight=asyncio%20coroutine#asyncio.coroutine) and removed in version 3.11. As a result, `glasgow` cannot be run:
```
  File "c:\users\maya\desktop\glasgow\software\glasgow\cli.py", line 31, in <module>
    from .applet import *
  File "c:\users\maya\desktop\glasgow\software\glasgow\applet\__init__.py", line 108, in <module>
    from ..access.simulation import *
  File "c:\users\maya\desktop\glasgow\software\glasgow\access\simulation\__init__.py", line 3, in <module>
    from .demultiplexer import *
  File "c:\users\maya\desktop\glasgow\software\glasgow\access\simulation\demultiplexer.py", line 13, in <module>
    class SimulationDemultiplexerInterface(AccessDemultiplexerInterface):
  File "c:\users\maya\desktop\glasgow\software\glasgow\access\simulation\demultiplexer.py", line 20, in SimulationDemultiplexerInterface
    @asyncio.coroutine
     ^^^^^^^^^^^^^^^^^
AttributeError: module 'asyncio' has no attribute 'coroutine'. Did you mean: 'coroutines'?
```

whitequark, 2022-12-17:
So this is kind of an issue because we rely on it for simulation. This probably needs some upstream Amaranth changes to retain simulation functionality. I guess for now we can require Python no newer than 3.10?..

sknebel, 2022-12-17:
There is also https://docs.python.org/3/library/types.html#types.coroutine to turn a generator into a coroutine. I'm not sure if it is 100% getting the same behavior, but seems at least similar?

whitequark, 2022-12-17:
Oh yeah we should look into that.

bkerler, 2023-03-29:
According to [here](https://stackoverflow.com/questions/39637675/what-is-the-difference-between-types-coroutine-and-asyncio-coroutine-decorator), types.coroutines is no replacement as it doesn't support yield. Instead asyncio could be made universal for anything >= python 3.6 but that would also require changes in amaranth as well as I've tested so far. The only alternative might be to use a custom class that separates python < 3.11 and python >= 3.11 support.

commit 0107d5f9 (2023-07-03): software: fix coroutines on Python 3.11

Fixes #297.