GlasgowEmbedded/glasgow #300: error: The 'amaranth' distribution was not found and is required by glasgow (install not working)

jeremyherbert, 2023-02-03:
Hi,

I'm trying to install the glasgow tools on a fresh install of ubuntu 22.04 (technically xubuntu). When I run the `setup.py` command as per the readme, I receive the following error:

```
error: The 'amaranth' distribution was not found and is required by glasgow
```

I have added the full log below:

```
$  python3 setup.py develop --user
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
  warnings.warn(
running develop
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
running egg_info
writing glasgow.egg-info/PKG-INFO
writing dependency_links to glasgow.egg-info/dependency_links.txt
writing entry points to glasgow.egg-info/entry_points.txt
writing requirements to glasgow.egg-info/requires.txt
writing top-level names to glasgow.egg-info/top_level.txt
writing manifest file 'glasgow.egg-info/SOURCES.txt'
running build_ext
Creating /home/jeremy/.local/lib/python3.10/site-packages/glasgow.egg-link (link to .)
glasgow 0.1.dev1687+ga283e8f is already the active version in easy-install.pth
Installing glasgow script to /home/jeremy/.local/bin

Installed /home/jeremy/Downloads/glasgow/software
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning:  is an invalid version and will not be supported in a future release
  warnings.warn(
Processing dependencies for glasgow==0.1.dev1687+ga283e8f
Searching for fx2>=0.9
Reading https://pypi.org/simple/fx2/
Downloading https://files.pythonhosted.org/packages/67/ac/2f5356ec2943753aad618f69eea72494da8b63ff87a8e2b624e95dfdfd07/fx2-0.9-py3-none-any.whl#sha256=25714295c826bdea3c7ce5f4c23f5f3be0145584c5d8aedc5554255e1d238bc3
Best match: fx2 0.9
Processing fx2-0.9-py3-none-any.whl
Installing fx2-0.9-py3-none-any.whl to /home/jeremy/.local/lib/python3.10/site-packages
Adding fx2 0.9 to easy-install.pth file
Installing fx2tool script to /home/jeremy/.local/bin

Installed /home/jeremy/.local/lib/python3.10/site-packages/fx2-0.9-py3.10.egg
Searching for amaranth
Doing git clone from https://github.com/amaranth-lang/amaranth.git to /tmp/easy_install-xn7hfw1e/amaranth.git
Best match: amaranth [unknown version]
Processing amaranth.git
Writing /tmp/easy_install-xn7hfw1e/amaranth.git/setup.cfg
Running setup.py -q bdist_egg --dist-dir /tmp/easy_install-xn7hfw1e/amaranth.git/egg-dist-tmp-0ej20173
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
warning: install_lib: 'build/lib' does not exist -- no Python modules to install

zip_safe flag not set; analyzing archive contents...
Moving UNKNOWN-0.4.dev58+ga704b1b-py3.10.egg to /home/jeremy/.local/lib/python3.10/site-packages
Adding UNKNOWN 0.4.dev58+ga704b1b to easy-install.pth file

Installed /home/jeremy/.local/lib/python3.10/site-packages/UNKNOWN-0.4.dev58+ga704b1b-py3.10.egg
error: The 'amaranth' distribution was not found and is required by glasgow
```

jeremyherbert, 2023-02-03:
If I install amaranth separately using `pip3 install --user --upgrade 'amaranth[builtin-yosys]'`, then the install succeeds:

```
$ python3 setup.py develop --user
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
  warnings.warn(
running develop
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
running egg_info
writing glasgow.egg-info/PKG-INFO
writing dependency_links to glasgow.egg-info/dependency_links.txt
writing entry points to glasgow.egg-info/entry_points.txt
writing requirements to glasgow.egg-info/requires.txt
writing top-level names to glasgow.egg-info/top_level.txt
writing manifest file 'glasgow.egg-info/SOURCES.txt'
running build_ext
Creating /home/jeremy/.local/lib/python3.10/site-packages/glasgow.egg-link (link to .)
glasgow 0.1.dev1687+ga283e8f is already the active version in easy-install.pth
Installing glasgow script to /home/jeremy/.local/bin

Installed /home/jeremy/Downloads/glasgow/software
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning:  is an invalid version and will not be supported in a future release
  warnings.warn(
Processing dependencies for glasgow==0.1.dev1687+ga283e8f
Searching for pyvcd==0.2.4
Best match: pyvcd 0.2.4
Adding pyvcd 0.2.4 to easy-install.pth file

Using /home/jeremy/.local/lib/python3.10/site-packages
Searching for libusb1==2.0.1
Best match: libusb1 2.0.1
Adding libusb1 2.0.1 to easy-install.pth file

Using /usr/lib/python3/dist-packages
Searching for fx2==0.9
Best match: fx2 0.9
Processing fx2-0.9-py3.10.egg
fx2 0.9 is already the active version in easy-install.pth
Installing fx2tool script to /home/jeremy/.local/bin

Using /home/jeremy/.local/lib/python3.10/site-packages/fx2-0.9-py3.10.egg
Searching for crcmod==1.7
Best match: crcmod 1.7
Adding crcmod 1.7 to easy-install.pth file

Using /usr/lib/python3/dist-packages
Searching for bitarray==1.6.3
Best match: bitarray 1.6.3
Adding bitarray 1.6.3 to easy-install.pth file

Using /usr/lib/python3/dist-packages
Searching for amaranth==0.3
Best match: amaranth 0.3
Adding amaranth 0.3 to easy-install.pth file
Installing amaranth-rpc script to /home/jeremy/.local/bin
Installing nmigen-rpc script to /home/jeremy/.local/bin

Using /home/jeremy/.local/lib/python3.10/site-packages
Searching for aiohttp==3.8.1
Best match: aiohttp 3.8.1
Adding aiohttp 3.8.1 to easy-install.pth file

Using /usr/lib/python3/dist-packages
Searching for Jinja2==2.11.3
Best match: Jinja2 2.11.3
Adding Jinja2 2.11.3 to easy-install.pth file

Using /home/jeremy/.local/lib/python3.10/site-packages
Searching for MarkupSafe==2.1.2
Best match: MarkupSafe 2.1.2
Adding MarkupSafe 2.1.2 to easy-install.pth file

Using /home/jeremy/.local/lib/python3.10/site-packages
Finished processing dependencies for glasgow==0.1.dev1687+ga283e8f
```

jeremyherbert, 2023-02-03:
Hmm, this may still be broken somehow, it seems that amaranth cannot find yosys:

```
$ glasgow run jtag-pinout --port A --pins-jtag 0,1,2,3 -M
I: g.device.hardware: building bitstream ID ea730248c9c25e0238b730da3a476f61
build_top.sh: 8: yosys: not found
Traceback (most recent call last):
  File "/home/jeremy/.local/bin/glasgow", line 33, in <module>
    sys.exit(load_entry_point('glasgow', 'console_scripts', 'glasgow')())
  File "/home/jeremy/Downloads/glasgow/software/glasgow/cli.py", line 857, in main
    exit(loop.run_until_complete(_main()))
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/home/jeremy/Downloads/glasgow/software/glasgow/cli.py", line 526, in _main
    await device.download_target(plan, rebuild=args.rebuild)
  File "/home/jeremy/Downloads/glasgow/software/glasgow/device/hardware.py", line 427, in download_target
    await self.download_bitstream(plan.execute(), plan.bitstream_id)
  File "/home/jeremy/Downloads/glasgow/software/glasgow/target/hardware.py", line 118, in execute
    products  = self.lower.execute_local(build_dir)
  File "/home/jeremy/.local/lib/python3.10/site-packages/amaranth/build/run.py", line 98, in execute_local
    subprocess.check_call(["sh", "{}.sh".format(self.script)])
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', 'build_top.sh']' returned non-zero exit status 127.
```

whitequark, 2023-02-03:
> When I run the `setup.py` command as per the readme, I receive the following error:
> 
> ```
> error: The 'amaranth' distribution was not found and is required by glasgow
> ```

I think what's happening here is that I converted amaranth-lang/amaranth to use pyproject.toml and Glasgow isn't there yet. I'll take a look.

> Hmm, this may still be broken somehow, it seems that amaranth cannot find yosys:

Glasgow requires two different versions of Yosys (one for Verilog export, one for synthesis); this is a different issue that I've filed as https://github.com/GlasgowEmbedded/glasgow/issues/302.

whitequark, 2023-02-07:
I think this should be fixed in 386a7b8fd5084f29da0d9a9b435cb7a53477ba45.

closed 2023-02-07 by whitequark