GlasgowEmbedded/glasgow #301: ImportError: cannot import name 'soft_unicode' from 'markupsafe'

jeremyherbert, 2023-02-03:
When running the `glasgow` executable for the first time after install, I get the following error:

```
$ glasgow
Traceback (most recent call last):
  File "/home/jeremy/.local/bin/glasgow", line 33, in <module>
    sys.exit(load_entry_point('glasgow', 'console_scripts', 'glasgow')())
  File "/home/jeremy/.local/bin/glasgow", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/jeremy/Downloads/glasgow/software/glasgow/cli.py", line 26, in <module>
    from .target.hardware import GlasgowHardwareTarget
  File "/home/jeremy/Downloads/glasgow/software/glasgow/target/hardware.py", line 8, in <module>
    from amaranth.build import ResourceError
  File "/home/jeremy/.local/lib/python3.10/site-packages/amaranth/build/__init__.py", line 3, in <module>
    from .plat import *
  File "/home/jeremy/.local/lib/python3.10/site-packages/amaranth/build/plat.py", line 6, in <module>
    import jinja2
  File "/home/jeremy/.local/lib/python3.10/site-packages/jinja2/__init__.py", line 12, in <module>
    from .environment import Environment
  File "/home/jeremy/.local/lib/python3.10/site-packages/jinja2/environment.py", line 25, in <module>
    from .defaults import BLOCK_END_STRING
  File "/home/jeremy/.local/lib/python3.10/site-packages/jinja2/defaults.py", line 3, in <module>
    from .filters import FILTERS as DEFAULT_FILTERS  # noqa: F401
  File "/home/jeremy/.local/lib/python3.10/site-packages/jinja2/filters.py", line 13, in <module>
    from markupsafe import soft_unicode
ImportError: cannot import name 'soft_unicode' from 'markupsafe' (/home/jeremy/.local/lib/python3.10/site-packages/markupsafe/__init__.py)
```

It appears that `markupsafe` needs to be pinned to 2.0.1 as they have removed this functionality: https://github.com/aws/aws-sam-cli/issues/3661

jeremyherbert, 2023-02-03:
After running `pip3 install --force-reinstall MarkupSafe==2.0.1` the `glasgow` command works.

whitequark, 2023-02-03:
This is an upstream issue, https://github.com/amaranth-lang/amaranth/issues/718.

whitequark, 2023-02-07:
I think this should be coincidentally fixed by 386a7b8fd5084f29da0d9a9b435cb7a53477ba45 since now pip will install the git revision of Amaranth.

closed 2023-02-07 by whitequark

daquintero, 2023-08-17:
Cheers for all the great work @whitequark ! Worth mentioning that the https://github.com/GlasgowEmbedded/glasgow/commit/386a7b8fd5084f29da0d9a9b435cb7a53477ba45 fix for the upstream pypi installation `amaranth @ git+https://github.com/amaranth-lang/amaranth.git` has not been propagated to pypi as it still is pinned to the old 0.3v without this fix.

whitequark, 2023-08-17:
> Worth mentioning that the [386a7b8](https://github.com/GlasgowEmbedded/glasgow/commit/386a7b8fd5084f29da0d9a9b435cb7a53477ba45) fix for the upstream pypi installation `amaranth @ git+https://github.com/amaranth-lang/amaranth.git` has not been propagated to pypi as it still is pinned to the old 0.3v without this fix.

I do not understand what you are saying here.

attie, 2023-08-20:
I think @daquintero is referring to [amaranth on PyPi](https://pypi.org/project/amaranth/#history).

@daquintero, it's worth noting, that the `name @ repo` syntax will install the package from the git repo, and not from PyPi, so while PyPi has v0.3 listed, it'll actually be installing from the [Amaranth repo](https://github.com/amaranth-lang/amaranth) directly.